PubInv/general-purpose-alarm-device #99: One of my board seems to intermittently fail on the write....

RobertLRead, 2022-10-27:
Possibly the other board fails as well. If this is not limited to one board we will have to investigate it as serious hardware defect.

ForrestErickson, 2022-10-28:
Regarding, "intermittently fail on the write.' and "Possibly the other board fails as well."

Please describe the symptoms of "fail" and describe the setup for how I might duplicate the failure.

RobertLRead, 2022-11-01:
At the time of uploading GPAD_API code, about one in 5 times it fails at the Arduino IDE "write flash" step (I think.) It produces an error message. Once this happens, I have to power down the unit and restart.

ForrestErickson, 2022-11-01:
Here is the error message when I did not press Reset timely.
[image]