PubInv/general-purpose-alarm-device #100: Develop A Plan To Build PolyVent SPI Controller Card Assemblies

ForrestErickson, 2022-10-28:
Now that there is a proposed PCB design, Lee proposed we build a few.

Identify the required quantity.  Lee suggests 5.
Choose to have JLCPCB build the SMT components or to build by hand. Lee is willing and suggest we build by hand however we might have more reliable assembly if done by JLCPCB.

We should order long lead parts. This time Lee has identified long lead parts in the BOM.
It is really the through hole three or four connectors and of course the PCB.

If accepted, this effort needs its own milestones.  Perhaps it should even be moved to a project of its own to reduce clutter of the GPAD development.

ForrestErickson, 2022-11-26:
On 20221115 Lee identifed long lead parts for @RobertLRead 
[image]

ForrestErickson, 2022-11-26:
@RobertLRead ,

Regarding Long Lead Parts Order. Have you placed the order below?
> [image]

While not yet reviewed, I have updated the PolyVent_SPI_Controller schematic and PCB per the review Lawrence and I had a couple of weeks ago. 
I have added a buck regulator which could more efficiently reduce the 24 volts to 12 volts for output to the GPAD.
> [image]

To test this voltage regulation configuration please order an additional item from Amazon: https://www.amazon.com/Zixtec-LM2596-Converter-Module-1-25V-30V/dp/B07VVXF7YX/ref=pd_lpo_2
For $15.98 we get ten of these useful regulators.
> 
[image]

Please advise.

FYI,
I hope to have one more review of the PCB and order assemblies from JLCPCB soon.

RobertLRead, 2022-12-02:
Yes, I just now ordered these. Thanks for documenting them so clearly in this issue.

ForrestErickson, 2022-12-06:
Hope to have parts in hand in Austin on 12 December. Robert will assemble.
From JLCPCB were ordered 5 raw PCBs and 5 assemblies with SMT.

closed 2022-12-06 by ForrestErickson