PubInv/general-purpose-alarm-device #104: Fully document screws for the case....

RobertLRead, 2022-10-28:
The Enclosure is working well, but to complete my verification of it, I tried to use screws on the sides which pin the whole thing together. I tried M1 and M1.75 screws and they are definitely too small. (Note: we cannot reach a nut on these screws.) M3 machine screws won't start.

I have ordered M2 screws to try them, but do not have any on hand. We may have to make the holes large and have them receive M3 screws. But maybe the M2 screws will work fine when i have them.

LokiMetaSmith, 2022-10-30:
the case screws, MACH SCREW PAN PHILLIPS #2-32 work well. These are apart of the digikey order we put in august. 
https://www.digikey.com/en/products/detail/pomona-electronics/4864/745092

These worked well for mounting the board in the 3D printed enclosure.  https://lcsc.com/product-detail/Screws_Shuntian-PA3X8_C357367.html
1.3mm Pointed Feet 5mm Self Tapping M3 8mm Fully Threaded Round head/Oval/Pan head Cross Slot Ni Plated 1010A 3x8mm Screws ROHS

RobertLRead, 2022-10-31:
M2 machine screws work.  Given that we are using M-X screws else, I think we should use M2 screws here.  Self-tapping screws are probably better here than machine screws, although I was able to get machine screws to work.

RobertLRead, 2022-11-01:
Lawrence, can you please add this to the BOM as a form of documentation and then close this. As a separate issue, where is the BoM as a GitHub file?

RobertLRead, 2022-11-01:
Note: This is related to: https://github.com/PubInv/general-alarm-device/issues/68

#68

LokiMetaSmith, 2022-11-21:
We need to adjust the part number from using machine screws to using sheet metal or plastic thread screws. It looks like 2mm is indeed the closest fit to  a #2 screw. 

[image]

ForrestErickson, 2022-11-21:
@LokiMetaSmith ,
Please suggest vendor(s) and part number(s).

And we need to put this information some where that someone trying to duplicate the whole GPAD will find it and know they have the full BOM for everything.

With our a research effort I do not fully know how to duplicate V1 PWA and enclosure right now.

LokiMetaSmith, 2022-11-22:
I believe this screw might be a good fit. 

The holes are suppose to be 2mm, so a 2.61mm thread dia. should be large enough to get good bite into the housing. The wall thickness is 2mm for each piece, so a total of 4mm of wall, where a 4mm screw will sit flush with the inner enclosure wall. 

https://www.mcmaster.com/99397A412/

LokiMetaSmith, 2022-11-22:
Should we add it to the schematic so it's captured in the project bom? 
we've partially done that with the screen and we have a listing for the enclosure. 
[image]

If so, it probably needs to be on it's own sheet and needs to include the screws for the case, and the screws for mounting the PCBA into the enclosure

ForrestErickson, 2022-12-15:
@LokiMetaSmith 

Would it make sense and do you have capacity to make up an OpenSCAD drawing with the screw mounting feature only on which to test driving the screws?

Something like 2"x2" filled with many mounting posts perhaps with different fits and test them for drive force and strip out torque?

Should we make an issue defining such an activity?