PubInv/general-purpose-alarm-device #107: GPAD With Bad Character

ForrestErickson, 2022-10-30:
Lee tested placing messages into alarms.
The GPAD responded with bad character (The opportunity to make puns here by all should be seized!)

**Notes on producing the error**
* The DUT (Device Under Test) is SN: 2
* Firmware is version: 0.1
* At the start of test, the DUT was in an unknown state after having probably been most recently exercised by the 'pheudoPolyVent
* Lee used RealTerm to send a few alarm commands to the DUT with out text message just to confirm normal opperation. 
* Lee then copyed the message from Robert's email "a3My pants are on fire" into RealTerm
* The GPAD displayed the message but Lee did not yet note if there was an error of Bad Character.
* Lee then started the endurance test and noted that when the a3 message was displayed it had the Bad Character. 
* Lee noted that the text echoed back out to the RealTerm did NOT have the Bad Character.

Screen Shot of RealTerminal. Note that the text pasted into the terminal is colored green. Text sent by typing into the RealTerminal terminal window is yellow as is text echoed from the DUT.
[image]

Text capture from the Real Terminal window. Note lots of white space which does not appear on the screen shot.

`I received: 2a2
                                                               
Command: a
                                                                    
2
                                                                             
currentlyMuted : 
                                                             
1
                                                                             
interpret Done
                                                                
Muted: YES
                                                                    
LVL: 2
                                                                        
No Message.
                                                                   

received: 1
                                                                  
bad format of command!
                                                        
PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
            
Muted: YES
                                                                    
LVL: 2
                                                                        
No Message.
                                                                   
I received: 1a
                                                                
Command: a
                                                                    
-48
                                                                           
Bad Level!
                                                                    
bad format of command!
                                                        
PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
            
currentlyMuted : 
                                                             
1
                                                                             
interpret Done
                                                                
Muted: YES
                                                                    
LVL: 2
                                                                        
No Message.
                                                                   

received: 21
                                                                 
bad format of command!
                                                        
PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
            
Muted: YES
                                                                    
LVL: 2
                                                                        
No Message.
                                                                   
a3My pants are on fireI received: 22a3My pants are on fire
                    
Command: a
                                                                    
3
                                                                             
currentlyMuted : 
                                                             
1
                                                                             
interpret Done
                                                                
My pants are on fire

                                                        
                                                                                
Muted: YES
                                                                    
LVL: 3
                                                                        
Msg: My pants are on fire
                                                     

received: 11Test Start
                                                       
bad format of command!
                                                        
PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
            
My pants are on fire

                                                        
                                                                                
Muted: YES
                                                                    
LVL: 3
                                                                        
Msg: My pants are on fire
                                                     

received: 2h
                                                                 
Command: h
                                                                    
PubInv GPAD: enter command in form CDa (C is a char, D is a digit)
            
currentlyMuted : 
                                                             
1
                                                                             
interpret Done
                                                                
My pants are on fire
                                                          
                      
                                                        
Muted: YES
                                                                    
LVL: 3
                                                                        
Msg: My pants are on fire
  
`

ForrestErickson, 2022-10-30:
Lee did not get a photo of the display showing the original Bad Character.  He thinks the Bad Character was the first character of "Msg"  The "M" looked like something from what might be an Asian character set.

However he was able to reproduce similar errors as photographed here.

[image]

[image]

Here is capture from RealTerminal after producing the last image.

`Muted: NO
                                                                     
LVL: 3
                                                                        
Msg: My pants are on fire
                                                     

received: 3a3
                                                                
Command: a
                                                                    
3
                                                                             
currentlyMuted : 
                                                             
0
                                                                             
interpret Done
                                                                

              
                                                               
Muted: NO
                                                                     
LVL: 3
                                                                        

g: 
                                                                          
a3My pants are on fireI received: 22a3My pants are on fire
                    
Command: a
                                                                    
3
                                                                             
currentlyMuted : 
                                                             
0
                                                                             
interpret Done
                                                                
My pants are on fire
                                                          
                      
                                                        
Muted: NO
                                                                     
LVL: 3
                                                                        
Msg: My pants are on fire
                                                     
I received: 2a3
                                                               
Command: a
                                                                    
3
                                                                             
currentlyMuted : 
                                                             
0
                                                                             
interpret Done
                                                                
Muted: NO
                                                                     
LVL: 3
                                                                        
No Message.
                                                                   
a3My pants are on fireI received: 22a3My pants are on fire
                    
Command: a
                                                                    
3
                                                                             
currentlyMuted : 
                                                             
0
                                                                             
interpret Done
                                                                
My pants are on fire
                                                          
                      
                                                        
Muted: NO
                                                                     
LVL: 3
                                                                        
Msg: My pants are on fire
   
`

ForrestErickson, 2022-10-30:
I found a specification for LCD controllers with a character set at Sparkfun.
Link: https://www.sparkfun.com/datasheets/LCD/HD44780.pdf

I have looked but do not see the character in question.
[image]

ForrestErickson, 2022-10-30:
20221030 10:35 Lee resumed testing.

Reset the DUT.
Pasted the message; 'a3My pants are on fire' into RealTerm

[image]
The display did NOT have the Bad Character.
[image]

Then typed "a3"
The display produced the Bad Character.
[image]

ForrestErickson, 2022-10-30:
Resume testing

Reset the DUT.
Setup RealTerm to send text strings WITH ONLY LF as a line termination.

[image]
The display had no Bad Character.

Change RealTerm to send with both CR and LF terminations.
[image]

The display produces the Bad Character.
[image]

Lee concludes that a CR (Carriage Return) character getting into the text string is likely the cause of the Bad Character.

ForrestErickson, 2022-10-30:
Lee has merged additional test files into the repository.
Update for GPAD_API_TEST with additional files and links for RealTerm… #108

In particular files with and with out CR (and LF) line terminations.

RobertLRead, 2023-01-31:
@ForrestErickson I think this is fixed in the latest version, please test.

ForrestErickson, 2023-06-20:
When using the GPAD_API_CONTROLER there are no unexpected characters.

closed 2023-06-20 by ForrestErickson