PubInv/general-purpose-alarm-device #116: Connect DTR on USB to Serial IC to The ATmega328 Reset Through Capacitor Like Arduino Nano Schematic

ForrestErickson, 2022-11-01:
This will bring to the GPAD the same program upload of a standard UNO / NANO.

From the Nano Schematic:
[image]

ForrestErickson, 2022-11-05:
Reference to CH340 data sheet: https://www.mpja.com/download/35227cpdata.pdf

Our GPAD uses the CH340E in the  MSOP-10 package. This part is the smallest but does not have the DTR pin.
The UNO in Lee's lab uses a CH340G which has the larger SOP-16 package and the DRT pin is on pin 13.

Reading the specification for the CH340 family the CH340G part has some EEPROM into which, for example, a serial number could be stored.  The CH340G is in the larger SOP-16 package only.

LokiMetaSmith, 2022-11-22:
Confirmed Resolved

closed 2022-11-22 by LokiMetaSmith