PubInv/general-purpose-alarm-device #117: Improve Holistic Documentation to be Reproducible by Outsider in Preparation of HardwareX Paper Submission

RobertLRead, 2022-11-01:
HardwareX requires that the "build instructions" for the devices it describe be completely reproducible. We need to improve our documentation to that level. In particular, the Enclosure at present may not be documented well enough.

LokiMetaSmith, 2022-11-21:
I think we need to decide if we're documenting V1 or V2 of the pcb, and which one we're submitting against, as that will change the documentation

ForrestErickson, 2022-11-21:
@LokiMetaSmith,
@RobertLRead ,

I think we should fully document V1 so that it is fully reproducible, warts and all.  Such documentation then forms a template for V2. After all, V2 does not yet exist and if we were to be run over by a bus before we bring it into existence then ....  

And I try to document as I go because I am no longer retaining details like my younger self did.

ForrestErickson, 2022-12-08:
20221208 Lee made descriptive changes to the file at: https://github.com/PubInv/general-alarm-device/blob/main/Hardware/Manufacturing/README.md

ForrestErickson, 2022-12-08:
20221208 Lee made changes to the main project README.md to better reflect our project status.