PubInv/general-purpose-alarm-device #118: Board SN #3 has a broken SPI RJ12 jack pointing upwards....

RobertLRead, 2022-11-01:
Here are the images:

[image]
[image]

RobertLRead, 2022-11-01:
Lee has photos which show this is shipping damage of some kind. (Or, maybe Rob broke when installing in enclosure, although he doubts it.)

ForrestErickson, 2022-11-01:
Image during Factory Test of #3 (left most of the four). Connector is normal.
[image]

ForrestErickson, 2022-11-01:
Notes on Lee's shipping to Robert.  Three GPAD PWAs were shipped in one box. Lee does not think it was the same box in which they were received. The three assemblies were individually padded but IIRC for two only with the same padding in which they were received and again IIRE, some alternate and supplemental padding bubble wrap was added.  Lee is rather sure they did not rattle around in the carton / box. 
Robert did not note any damage to the carton / box.

While this may be a one off and this is damage while shipping a unit with out an enclosure, this reminds us that we should identify a card board box and appropriate padding in which to ship an individual unit.

ForrestErickson, 2022-11-01:
As this is a one off I am closing. I will create an issue to identify a carton and packing in which a single GPAD in an enclosure can be shipped.

ForrestErickson, 2022-11-15:
Closing as won't fix as we can do nothing further.

closed 2023-01-04 by LokiMetaSmith