PubInv/general-purpose-alarm-device #119: Identify Carton and Padding For Shipping Finished GPAD Product

ForrestErickson, 2022-11-01:
Assuming the enclosure is about 6x4x 2 inches, a ULine model S-4080 carton might be good.  See:

https://www.uline.com/Product/Detail/S-4080/Corrugated-Boxes-200-Test/8-x-6-x-4-Corrugated-Boxes
[image]

There will need to be bubble wrap to fill out the space.   The 1/2" Dunnage [S-3930P](https://www.uline.com/Product/Detail/S-3930P/Bubble-Rolls/Economy-Bubble-Roll-12-x-250-1-2-Perforated) looks good.
https://www.uline.com/BL_470/Uline-Economy-Air-Bubble
[image]

LokiMetaSmith, 2022-11-21:
I have issues with bubble wrap from a sustainability and ESD dissipative angle. I would prefer an esd baggie and paper infill. 

also, is this for a fully assembled case with board, or just shipping the pcb's? As shipped from jlcpcb, they use pink foam between boards and the whole thing wrapped in pink  bubble wrap, Pink plastic/bubble wrap is anti static forming and slightly dissipative.

ForrestErickson, 2022-11-21:
Regarding, "would prefer an esd baggie and paper infill."  @LokiMetaSmith  Please suggest some vendor(s) and part number(s)

This is for shipping a full finished product enclosure, instructions and everything. 

During development we have to improvise for the case at hand.  Unfortunately I have lots of pink bubble wrap I have received products in for years and when I ship it away at least gets used twice.

ForrestErickson, 2022-12-21:
@LokiMetaSmith 
Regarding, ". I would prefer an esd baggie and paper infill."

Please suggest exact vendor and vendor part numbers.