PubInv/general-purpose-alarm-device #121: Git (and presumably Github) Changes Line Ending on Files. Invalidates Testing

ForrestErickson, 2022-11-02:
On 20221102, while making the GPAD test file GPAD_API_TEST_ALPHANUM1.txt and editing the file GPAD_API_TEST_3LF_Only.txt I examined them to discover what had been the single LF character had become CR LF.   It may be that perhaps Notepad++ makes that change.

However I see that for sure when such a file is added to version control in Git that Git makes such a change.

[image]

This means that the test files in the folder at: 
https://github.com/PubInv/general-alarm-device/tree/main/Firmware/GPAD_API_TEST
Cannot be trusted with out review. 

Tests with them may be invalid.

Perhaps we should zip up the files and commit them as a work around.

ForrestErickson, 2022-11-05:
There is an argument for git to control line ending conversion.

https://git-scm.com/docs/gitattributes

[image]

Example:  *.sh		text eol=lf

ForrestErickson, 2022-11-09:
ON 20221109 Lee added a gitattribute file to control line endings
> # Attributes to control line endings for tsting GPAD_API

>  *.lf.txt		text eol=lf
> *.cr.txt		text eol=cr
> *.crLF.txt		text eol=crlf

> *.LF.txt		text eol=lf
> *.CR.txt		text eol=cr
> *.CRLF.txt		text eol=crlf

After changing file names to reflect the desired line endings and again modifying the files this status message was produced:
[image]

LokiMetaSmith, 2022-11-21:
the great line ending delimitator debate strikes again. 

the test code should be  able to handle all cases.  but it's good to be aware of it, as I generally ignore that warning and it's subtle implications.

LokiMetaSmith, 2022-11-22:
Confirmed Resolved

closed 2022-11-22 by LokiMetaSmith