PubInv/general-purpose-alarm-device #122: GPAD_API Version 0.1 Modifies Text When Echoed, Unexpected or Undocumented Behavior.

ForrestErickson, 2022-11-02:
Testing the GPAD_API Version 0.1 with the new file GPAD_API_TEST_ALPHANUM1.txt  
* [image]

There is Unexpected or Undocumented Behavior to the text traffic on the serial port.

Lee noticed that the DUT echoed back in the terminal window altered strings.  Screen shot of RealTerminal testing:
* [image]

For example the text "a3THE QUICK BROWN FOX JUMPED OVER THE LAZY DOG." was echoed as 

```
I received: 47a3THE QUICK BROWN FOX JUMPED OVER THE LAZY DOG.
                 
Command: a
                                                                    
3
                                                                             
currentlyMuted : 
                                                             
0
                                                                             
interpret Done
                                                                
THE QUICK BROWN FOX 
                                                          
JUMPED OVER THE LAZY
                                                          
 DOG.
 

```
which has additional CRLF inserted.  This mark down document renders the CR and LF into the text displayed just above this note.

Pasting into the RealTerm terminal window (the green text in this screen shot) produces similar results.
* [image]

Pasting into the Arduino terminal window produces similar results but with out the helpful indication of CR and LF which RealTerminal provides.
*[image]

ForrestErickson, 2022-11-02:
**Caution** on using the file GPAD_API_TEST_ALPHANUM1.txt from the repository. Git changes LF to CRLF and so the file you pull out from the link at: https://github.com/PubInv/general-alarm-device/tree/main/Firmware/GPAD_API_TEST is today 20221102 10:40 AM corrupted.  There is an issue about this corruption.

RobertLRead, 2023-01-31:
Fixed by GIT configuration.

closed 2023-01-31 by RobertLRead