PubInv/general-purpose-alarm-device #127: Board SN5 gives error verifying upload, even though upload works....

RobertLRead, 2022-11-11:
Using the board SN5, I can upload a program, but get the following errors when we try to upload.

[image]

ForrestErickson, 2022-11-12:
@RobertLRead 
I will do some googeling and try to learn more.
The screen capture of the error is helpful.
It would help further if you could additionally copy the text of the error. In the IDE press the "copy error message" button which does so.

Example of screen shot.
> [image]

Example of the captured text from the IDE

> Arduino: 1.8.19 (Windows 10), Board: "Arduino Duemilanove or Diecimila, ATmega328P"
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> GPAD_Factory_Test:40:1: error: expected ',' or ';' before 'const'
> 
>  const int LOW_TIME_LED_MS = 200;
> 
>  ^~~~~
> 
> C:\Users\Public\Downloads\Arduino\PubInv\general-alarm-device\Firmware\GPAD_Factory_Test\GPAD_Factory_Test.ino: In function 'void updateWink()':
> 
> GPAD_Factory_Test:90:27: error: 'LOW_TIME_LED_MS' was not declared in this scope
> 
>        nextLEDchangee_ms = LOW_TIME_LED_MS;
> 
>                            ^~~~~~~~~~~~~~~
> 
> C:\Users\Public\Downloads\Arduino\PubInv\general-alarm-device\Firmware\GPAD_Factory_Test\GPAD_Factory_Test.ino:90:27: note: suggested alternative: 'HIGH_TIME_LED_MS'
> 
>        nextLEDchangee_ms = LOW_TIME_LED_MS;
> 
>                            ^~~~~~~~~~~~~~~
> 
>                            HIGH_TIME_LED_MS
> 
> exit status 1
> 
> expected ',' or ';' before 'const'
> 
> 
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
>

ForrestErickson, 2022-11-12:
@RobertLRead 

Some googeling on "stk5--_paged_load(): (a) protocol error, expect-0x10, resp-0x67" I found some traffic and I learned that there is a more verbose mode of error reporting from the Arduino IDE.
See screen shot.
[image]

ForrestErickson, 2022-11-12:
Notes on "stk5--_paged_load(): (a) protocol error, expect-0x10, resp-0x67" 
Found similar error here: https://arduino.stackexchange.com/questions/23366/avrdude-stk500-protocol-error where the at least one user was messing with the loading the target with the "Arduino as ISP".

Here on a similar error I found traffic indicating sensitivity to the version of the IDE back in 2019.
https://forum.arduino.cc/t/avrdude-stk500_disable-protocol-error-expect-0x14-resp-0x10/590824

Lee is using Arduino IDE version 1.8.19.

@RobertLRead what is your version IDE?

**All this aside this might be yet another way that the missing reset from the USB interface to the ATmega328 where we have to press the reset on the DUT may manifest**

RobertLRead, 2022-11-30:
I am using 1.8.19.

Here is the example text:

Arduino: 1.8.19 (Mac OS X), Board: "Arduino Duemilanove or Diecimila, ATmega328P"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/robertread/Library/Arduino15/packages -hardware /Users/robertread/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/robertread/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/robertread/Documents/Arduino/libraries -fqbn=arduino:avr:diecimila:cpu=atmega328 -vid-pid=0000_0000 -ide-version=10819 -build-path /var/folders/1x/lxn7mlf54h382s81tr8z06tm0000gn/T/arduino_build_29827 -warnings=default -build-cache /var/folders/1x/lxn7mlf54h382s81tr8z06tm0000gn/T/arduino_cache_494818 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/robertread/PubInv/general-alarm-device/Firmware/GPAD_API_SPI_CONTROLLER/GPAD_API_SPI_CONTROLLER.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/robertread/Library/Arduino15/packages -hardware /Users/robertread/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/robertread/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/robertread/Documents/Arduino/libraries -fqbn=arduino:avr:diecimila:cpu=atmega328 -vid-pid=0000_0000 -ide-version=10819 -build-path /var/folders/1x/lxn7mlf54h382s81tr8z06tm0000gn/T/arduino_build_29827 -warnings=default -build-cache /var/folders/1x/lxn7mlf54h382s81tr8z06tm0000gn/T/arduino_cache_494818 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/robertread/PubInv/general-alarm-device/Firmware/GPAD_API_SPI_CONTROLLER/GPAD_API_SPI_CONTROLLER.ino
Using board 'diecimila' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Warning: Board breadboard:avr:atmega328bb doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA328BB
Detecting libraries used...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard /var/folders/1x/lxn7mlf54h382s81tr8z06tm0000gn/T/arduino_build_29827/sketch/GPAD_API_SPI_CONTROLLER.ino.cpp -o /dev/null
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src /var/folders/1x/lxn7mlf54h382s81tr8z06tm0000gn/T/arduino_build_29827/sketch/GPAD_API_SPI_CONTROLLER.ino.cpp -o /dev/null
Alternatives for SPITransfer.h: [SerialTransfer@3.1.3]
ResolveLibrary(SPITransfer.h)
  -> candidates: [SerialTransfer@3.1.3]
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR -I/Applications/A [... cut ...]
avrdude: input file /var/folders/1x/lxn7mlf54h382s81tr8z06tm0000gn/T/arduino_build_29827/GPAD_API_SPI_CONTROLLER.ino.hex contains 4766 bytes
avrdude: reading on-chip flash data:

Reading | 
avrdude: stk500_paged_load(): (a) protocol error, expect=0x10, resp=0xee
#avrdude: stk500_cmd(): programmer is out of sync
avr_read(): error reading address 0x0000
    read operation not supported for memory "flash"
avrdude: failed to read all of flash memory, rc=-2
avrdude: stk500_disable(): protocol error, expect=0x14, resp=0xc3

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected

ForrestErickson, 2022-11-30:
Lee's notes on a normal upload on Windows platform.

> C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Admin\Documents\Arduino\libraries -fqbn=arduino:avr:diecimila:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10819 -build-path C:\Users\Admin\AppData\Local\Temp\arduino_build_502233 -warnings=none -build-cache C:\Users\Admin\AppData\Local\Temp\arduino_cache_928674 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Public\Downloads\Arduino\PubInv\general-alarm-device\Firmware\GPAD_API\GPAD_API.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Admin\Documents\Arduino\libraries -fqbn=arduino:avr:diecimila:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10819 -build-path C:\Users\Admin\AppData\Local\Temp\arduino_build_502233 -warnings=none -build-cache C:\Users\Admin\AppData\Local\Temp\arduino_cache_928674 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Public\Downloads\Arduino\PubInv\general-alarm-device\Firmware\GPAD_API\GPAD_API.ino
Using board 'diecimila' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_502233\\sketch\\GPAD_API.ino.cpp" -o nul
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src" "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_502233\\sketch\\GPAD_API.ino.cpp" -o nul
Using cached library dependencies for file: C:\Users\Admin\AppData\Local\Temp\arduino_build_502233\sketch\alarm_api.cpp
Using cached library dependencies for file: C:\Users\Admin\AppData\Local\Temp\arduino_build_502233\sketch\gpad_serial.cpp
Using cached library dependencies for file: C:\Users\Admin\AppData\Local\Temp\arduino_build_502233\sketch\gpad_utility.cpp
Using cached library dependencies for file: C:\Users\Admin\AppData\Local\Temp\arduino_build_502233\sketch\robot_api.cpp
Alternatives for DailyStruggleButton.h: [DailyStruggleButton@0.5.1]
ResolveLibrary(DailyStruggleButton.h)
  -> candidates: [DailyStruggleButton@0.5.1]
Alternatives for LiquidCrystal_I2C.h: [LCD_I2C-master@1.1.4]
ResolveLibrary(LiquidCrystal_I2C.h)
  -> candidates: [LCD_I2C-master@1.1.4]
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src\SPI.cpp
Using cached library dependencies for file: C:\Users\Admin\Documents\Arduino\libraries\DailyStruggleButton\src\DailyStruggleButton.cpp
Using cached library dependencies for file: C:\Users\Admin\Documents\Arduino\libraries\LCD_I2C-master\LiquidCrystal_I2C.cpp
[... cut ...]
avrdude: verifying flash memory against C:\Users\Admin\AppData\Local\Temp\arduino_build_502233/GPAD_API.ino.hex:
avrdude: load data flash data from input file C:\Users\Admin\AppData\Local\Temp\arduino_build_502233/GPAD_API.ino.hex:
avrdude: input file C:\Users\Admin\AppData\Local\Temp\arduino_build_502233/GPAD_API.ino.hex contains 9500 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 2.05s

avrdude: verifying ...
avrdude: 9500 bytes of flash verified

avrdude done.  Thank you.

ForrestErickson, 2022-11-30:
Note that the program sequence captured by Robert includes 

> avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "/var/folders/1x/lxn7mlf54h382s81tr8z06tm0000gn/T/arduino_build_29827/GPAD_API_SPI_CONTROLLER.ino.hex"
avrdude: writing flash (4766 bytes):

And by Lee 

> Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\Admin\AppData\Local\Temp\arduino_build_502233/GPAD_API.ino.hex"
avrdude: writing flash (9500 bytes):

So on the failed device SN5 the target hardware was in communication at that point in the programing cycle. 

The final failure  on SN5 reported was:
> Reading |
avrdude: stk500_paged_load(): (a) protocol error, expect=0x10, resp=0xee
#avrdude: stk500_cmd(): programmer is out of sync
avr_read(): error reading address 0x0000
read operation not supported for memory "flash"
avrdude: failed to read all of flash memory, rc=-2
avrdude: stk500_disable(): protocol error, expect=0x14, resp=0xc3

avrdude done. Thank you.

the selected serial port
does not exist or your board is not connected

Lee guesses that the part of the message "the selected serial port does not exist or your board is not connected" is an inference made because of the failure " Reading |avrdude: stk500_paged_load(): (a) protocol error"

This indicated that during the programing of the target the read operation failed.  I guess time to learn more about "Reading |avrdude: stk500_paged_load():"

ForrestErickson, 2022-11-30:
FYI, I have made some additional notes in the folder: https://github.com/PubInv/general-alarm-device/blob/main/Hardware/Documentation/TroubleShooting/SN00005/Troubleshooting_SN00005.md

These notes include Lee figuring out how to use avrdude to verify only.

Bottom line on my windows maching I had to open a cmd window and navigate to the folder with avrdude in it to run it.
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin>

Then I composed the following command line argument:
-C../etc/avrdude.conf -v -patmega328p -carduino -PCOM16 -b57600 -D -Uflash:v:C:\Users\Admin\AppData\Local\Temp\arduino_build_502233/GPAD_API.ino.hex:i

This produced a verify only action on the GPAD on which I am experimenting SN2. SN2 has the rework for DTR to Reset so that the device resets under the command of the USB/UART/

> avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "../etc/avrdude.conf"

         Using Port                    : COM16
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us
         
         avrdude: AVR device initialized and ready to accept instructions
         Reading | ################################################## | 100% 0.01s
         avrdude: Device signature = 0x1e950f (probably m328p)
         avrdude: safemode: lfuse reads as 0
         avrdude: safemode: hfuse reads as 0
         avrdude: safemode: efuse reads as 0
         avrdude: verifying flash memory against C:\Users\Admin\AppData\Local\Temp\arduino_build_502233/GPAD_API.ino.hex:
         avrdude: load data flash data from input file C:\Users\Admin\AppData\Local\Temp\arduino_build_502233/GPAD_API.ino.hex:
         avrdude: input file C:\Users\Admin\AppData\Local\Temp\arduino_build_502233/GPAD_API.ino.hex contains 9500 bytes
         avrdude: reading on-chip flash data:
         
         Reading | ################################################## | 100% 2.49s
         
         avrdude: verifying ...
         avrdude: 9500 bytes of flash verified
         avrdude: safemode: lfuse reads as 0
         avrdude: safemode: hfuse reads as 0
         avrdude: safemode: efuse reads as 0
         avrdude: safemode: Fuses OK (E:00, H:00, L:00)
         
         avrdude done.  Thank you.
         
         C:\Program Files (x86)\Arduino\hardware\tools\avr\bin>

ForrestErickson, 2022-11-30:
So SN5 might be probed by trying to do the verify only step at a command line.

Use avrdude -C../etc/avrdude.conf -v -patmega328p -carduino -PCOM15 -b57600 -D -Uflash:v:C:\Users\Admin\AppData\Local\Temp\arduino_build_502233/GPAD_API.ino.hex:i 
However change the port from "COM15" to the appropriate port for your system and device.

I was able to get this to work even on SN6 which does not have the rework for DTR to Reset by pressing the reset button immediate after entering the command above.

Screen shot of succsfull verify.
[image]

@RobertLRead 
If you have time please try to preform this command line verify of the SN5. 
It would be of interest to know if SN5 ever verifies. Does it fail every single time or is it intermittent?

ForrestErickson, 2023-01-02:
On 20230102 Lee had similar error on serial number 2.

[image]

> Arduino: 1.8.19 (Windows 10), Board: "Arduino Duemilanove or Diecimila, ATmega328P"
> 
> Sketch uses 4772 bytes (15%) of program storage space. Maximum is 30720 bytes.
> 
> Global variables use 341 bytes (16%) of dynamic memory, leaving 1707 bytes for local variables. Maximum is 2048 bytes.
> 
> 
> 
> avrdude: stk500_paged_load(): (a) protocol error, expect=0x10, resp=0x08
> 
> avrdude: stk500_cmd(): programmer is out of sync
> 
> avr_read(): error reading address 0x0000
> 
>     read operation not supported for memory "flash"
> 
> avrdude: failed to read all of flash memory, rc=-2
> 
> avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x08
> 
> avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x08
> 
> 
> 
> 
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
>