PubInv/general-purpose-alarm-device #133: Switch S601 Foot Print Needs Holes for Locating Pins

ForrestErickson, 2022-11-13:
For Version 1 we had to cut off the locating pins.
Update PCB Footprint for holes.

ForrestErickson, 2022-11-13:
Detail from data sheet:
[image]

ForrestErickson, 2022-11-13:
Update footprint and import to PCB Done.
[image]

ForrestErickson, 2022-11-13:
@LokiMetaSmith 
This and the fixed footprint for the I2C expander chip, U301 are updated.

If you could print out the PCB 1:1 and test fit with the parts I assume you have that would act as a sanity check.

Lee

LokiMetaSmith, 2022-11-21:
I'll do this as part of the Version 2 evaluation