PubInv/general-purpose-alarm-device #137: Investigate Louder Buzzer for GPAD.

ForrestErickson, 2022-11-15:
Increase voltage drive to the existing buzzer

Investigate other buzzers.

ForrestErickson, 2023-01-07:
This speaker might also be a solution: 

> 
[image]

LINK:
https://www.amazon.com/Yootop-Internal-Magnet-Loudspeaker-Speaker/dp/B07FMR5JGX/ref=sr_1_3

RobertLRead, 2023-01-07:
That does look pretty good!

ForrestErickson, 2023-01-07:
Here is one rated at 2W.
https://www.amazon.com/Uxcell-a15080600ux0275-Internal-Magnet-Speaker/dp/B0177ABRQ6/ref=pd_rhf_d_se_s_pd_sbs_rvi_sccl_2_5/142-6809396-6319936?

> 
[image]

RobertLRead, 2023-01-31:
Lawrence has received and is building a case around this.

closed 2023-01-31 by RobertLRead