PubInv/general-purpose-alarm-device #138: Make Photo of Rev2 Enclosure with GPAD Light up and Playing for our Home Page

ForrestErickson, 2022-11-15:


RobertLRead, 2022-11-15:
Here is the photo I took:

[image]

ForrestErickson, 2022-11-15:
@RobertLRead ,
@LokiMetaSmith 

I note how the LEDs are resessed below the enclosure.

I left some additional lead length Perhaps they could be moved up another 3/16 or more.

We need assembly instructions on setting the LED height at solder time.

RobertLRead, 2022-11-15:
Yes, this is an issue---in fact, I had to adjust them a bit by hand
(bending them to line up.)

The simplest thing is to just make the leads a little longer and possibly
make the holes for them slightly larger to be sure that the lens of the LED
can fit inside them

LokiMetaSmith, 2022-11-21:
The screen is the tallest part of the device, and I can increase the size of the holes pretty easily so the LED's protrude. I think we should fix that as a feature, and I can print out a quick slotted block that holds the led's in place. Technically, they make such things, but when you have a 3D printer, everything looks like a solution looking for another 3D print. 

can we get a side profile view of the connectors, a top view of the screen? 

What I see working is the speaker holes, the button height could come down a little bit (question for @RobertLRead ), and it's hard to tell, but is the screen perfectly aligned? it looks like it's offset and needs to come down a bit.

RobertLRead, 2022-11-23:
It appears to me that the screen is about 1mm hither than the orifice (that is, the orifice could come down a bit.) @LokiMetaSmith  

For the photos you requested, please see: [](https://github.com/PubInv/general-alarm-device/issues/146)

ForrestErickson, 2022-11-23:
Lee added GPAD in Prototype Enclosure image to the main README.md at the GPAD root.

closed 2022-11-23 by ForrestErickson