PubInv/general-purpose-alarm-device #140: Identify Selector Switch for nCS, PolyVent_SPI_Controller

ForrestErickson, 2022-11-15:
During the discussion of the PolyVent_SPI_Controller PCB it was suggested that we have a rotary selection switch.

Please @LokiMetaSmith , @RobertLRead help identify a desired switch for Lee

LokiMetaSmith, 2022-11-21:
I think the dip switch is the best option. or jumper pins

I was also thinking of rotary encoders, but I forgot that most of them are binary encoders, and without a demultiplexor it wouldn't work.

[image]

https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=binary

https://jlcpcb.com/partdetail/TexasInstruments-CD74HC238PWR/C403724

ForrestErickson, 2022-11-24:
I have updated the Schematic and PCB with these changes and made a PDF of schematic. Ready for review.

closed 2022-12-06 by ForrestErickson