PubInv/general-purpose-alarm-device #141: Need Assembly Instructions on Setting the LED Height at Solder Time.  Version ?2? Enclosure

ForrestErickson, 2022-11-15:
The Version ?2? enclosure at @RobertLRead  location when assembled with the Version 1 PCB with LEDs set for the same height as the LCD has zero LED protrusion.

Perhaps could be set higher.
Perhaps a spacer of some sort to set height.
Perhaps a fixture to holed at the correct height while soldering.
Perhaps some gauge for go-nogo on the height would help the operator.

@LokiMetaSmith did you have further ideas / intentions on making a spacer for the LEDs?

LokiMetaSmith, 2022-11-22:
It'll need a plastic spacer to hold the LED's in place. Without the spacer, the risk of misalignment on long leaded LED's is too high.

LokiMetaSmith, 2023-06-20:
Spacers  
https://github.com/PubInv/general-purpose-alarm-device/blob/main/Hardware/Enclosure/U_Box_V105_GPAD_LED_Standoff_single.stl

closed 2023-06-20 by LokiMetaSmith