PubInv/general-purpose-alarm-device #143: Add a "smoke alarm test mode"......

RobertLRead, 2022-11-17:
Although it will alway be obvious if their is power to a GPAD or not due to the "heartbeat" LED, it would be nice to be able to demo features WITHOUT having to interface.

I propose that if you push AND HOLD the mute button for 2 seconds or more, it enter increasing urgently alarm states until the button is released. This lets you test the light and sound with only power (no data connection!)

This can be accomplished entirely in firmware, so I am assigning it to myself.

ForrestErickson, 2022-11-17:
Regarding, "demo features WITHOUT having to interface."
There are horror stories about medical devices with demo mode from which we should learn.

My memory must NOT be relied upon but here goes.

1. DEMO MODE must clearly be indicated. 
2. The device must exit the demo mode with out any password or other heroic effort on the part of the user. 
3. The device had better not power up in demo mode.

Regarding demonstrating the GPAD.  I was considering how an ESP32 attached to a Pulse Oximeter and with the GPAD compatible SPI interface would make a fun medical demonstration use case.  
See: https://www.amazon.com/AITRIP-MAX30102-Breakout-Compatible-Raspberry/dp/B08ZS5CYSJ?th=1

Lee

ForrestErickson, 2023-01-02:
Perhaps you could enter the demo alarm mode by checking for the Mute button to be pressed (closed) on power up.  

This would have the added consequence that if the button is stuck on the GPAD would alert the user on power up of something wrong. 

The possible down side is that it would REQUIRE power cycling with the button open to clear the mode unless some time out or other user pressing the button and maybe navigating a user interface was implemented.