PubInv/general-purpose-alarm-device #145: Determine values for R306 and R308

LokiMetaSmith, 2022-11-22:
[image]

We have enough boards in hand that we can measure the resistor value needed to populate R306 and R308, which would make the potentiometer RV301 redundant. 

This should be done to remove a hand assembly element to the board, with the assumption that the displays used are consistent in their contrast ratio

ForrestErickson, 2022-11-22:
RV301 sets the voltage into the LCD called V0
Regarding setting V0 with fixed resistors.
During the assembly of five units in Maryville I captured the voltage to which I set the DUTs by serial number into a table in the document: https://github.com/PubInv/general-alarm-device/blob/main/Hardware/Documentation/ManufacturingUnitTestTroubleshooting.md
See table below.
I did not evaluate the rage of voltages to which I could set the display for "acceptable contrast" nor have I explored defining "acceptable contrast" which I know must include off axis viewing angle on a cone (A quick sample size one check shows that the characters become invisible at what I am guessing is greater than 60 degrees off axis but less than 80 degrees..  Nor have I explored temperature variation.

We should note which vendor we are testing. IIRC we have displays from two different distributors but we do not know the ultimate manufacturer.

If I had willing technical help. I would suggest characterizing these unknowns and only then making a decision on fixed resistors.   

Other factors. 
The reoccurring part cost of the Bourns potentiometer is listed as 0.6963.
The reoccurring cost to adjust in the factory test is TBD but it takes turning a screw driver and I always did it with a multi meter so that I could watch the multimeter and aim for a target voltage similar to previous displays.  Let's assume 3 minuets.
Last time I worked for compensation it was at about 0.64 per minuet so my labor cost would be $1.92
Total material and labor cost might be about $2.62.  The cost for fixed resistors will be negligible.

So how many units would we have to build to break even to recover the engineering cost to determine if we can choose fixed resistors and testing our assumption. 
If it takes one day the engineering cost might be something like 8 * 100,000/2080 = $384.  So less than 100 units for break even.
If it takes one week the engineering cost might be something like 40 * 100,000/2080 = $1923.  So about 733 units for break even

That was a fun "what if?"

Table screen shot:
> [image]

ForrestErickson, 2022-12-14:
@LokiMetaSmith  I just assigned you and me to this issue.
As you can see from my notes above I am hoping you capture the V0 voltage which produces good contrast on the remaining units and for each vendor of LCD so that we can learn as much as we can about device to device variation and vendor to vendor variation for insight into this decision.

If we use 5% resistors then we need to verify that with a 5% variation from the nominal Vo that all units have good contrast. And of course I have not put much thought into defining "good" contrast such that it is a measurement. (We physicists like measurements.)

If we had a priority ranking on Issues I would lobby for this to be of the lowest priority.

closed 2023-01-31 by RobertLRead