PubInv/general-purpose-alarm-device #146: Case orifice alignment is almost but not quite perfect...

RobertLRead, 2022-11-22:
Here are some photos of the case. Everything works, but some minor adjustments might be warranted:

[image]
[image]

ForrestErickson, 2022-11-23:
@RobertLRead 
I suggest a change to the issue name.
From: Case orifice alignment is almost but not quite perfect...#146
To: Enclosure Prototype #1, Connector and LED Fit Notes

RobertLRead, 2022-11-23:
I've changed the name.  It is a low priority for me to try to fix the leads at present. The nail polish idea is a nice idea--Of course, a colored plastic would also affect this, so I consider this much lower priority than the software I am working on.

ForrestErickson, 2022-11-23:
Lee added GPAD in Prototype Enclosure image to the main README.md at the GPAD root.

LokiMetaSmith, 2022-12-27:
1. Enclosure USB header slot needs to be moved down towards the board 1mm, and 1mm away from the center of the board. 
2. The latch on the i2c header needs to be evaluated, in terms of interference with the enclosure. This will require making a test cable.  
3. LED holes need to be expanded to allow LED's protruding

ForrestErickson, 2023-06-20:
The files with name/ number V105 meet these requirements.