PubInv/general-purpose-alarm-device #149: BOM and Pick And Place To Have JPLPCB PolyVent SPI Controller

ForrestErickson, 2022-11-28:
For the PolyVent SPI Controller if JPLPCB is to build the board and place the components they can, we need a BOM and a Pick and Place file that meets the JCLPCB requirements.  From the example files at JLCPCB we have a BOM example and a placement example

### BOM example
<HTML>
<BODY TEXT="#000000">
<TABLE FRAME=VOID CELLSPACING=0 COLS=4 RULES=NONE BORDER=0>
	<COLGROUP><COL WIDTH=191><COL WIDTH=91><COL WIDTH=265><COL WIDTH=179></COLGROUP>
	<TBODY>
		<TR>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" WIDTH=191 HEIGHT=27 ALIGN=CENTER VALIGN=MIDDLE SDNUM="1033;1033;General"><B><FONT SIZE=3 COLOR="#000000">Comment</FONT></B></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" WIDTH=91 ALIGN=CENTER VALIGN=MIDDLE SDNUM="1033;1033;General"><B><FONT SIZE=3 COLOR="#000000">Designator</FONT></B></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" WIDTH=265 ALIGN=CENTER VALIGN=MIDDLE SDNUM="1033;1033;General"><B><FONT SIZE=3 COLOR="#000000">Footprint</FONT></B></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" WIDTH=179 ALIGN=CENTER VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT FACE="宋体" COLOR="#AFABAB">JLCPCB Part #（optional）</FONT></TD>
		</TR>
		<TR>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" HEIGHT=27 ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">100uF</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">C1</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">CAP-SMD_L3.5-W2.8</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">C16133</FONT></TD>
		</TR>
		<TR>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" HEIGHT=27 ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">15K</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">R5,R6</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">R0603</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">C22809</FONT></TD>
		</TR>
		<TR>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" HEIGHT=27 ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">DC-005-20A_C136744</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">DC1</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">DC-IN-TH_DC-005-20A</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">C136744</FONT></TD>
		</TR>
		<TR>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" HEIGHT=27 ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">Header-Female-2.54_1x8</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">P2,P3</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">HDR-TH_8P-P2.54-V</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">C27438</FONT></TD>
		</TR>
		<TR>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" HEIGHT=27 ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">K2-1102DP-C4SW-04</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">SW1</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">KEY-TH_4P-L6.0-W6.0-P4.50-LS6.5</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">C110153</FONT></TD>
		</TR>
		<TR>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" HEIGHT=27 ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">ATMEGA328P-PU</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">U1</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">DIP-28_L34.6-W7.3-P2.54-LS10.2-BL</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">C33901</FONT></TD>
		</TR>
		<TR>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" HEIGHT=27 ALIGN=LEFT VALIGN=MIDDLE SDVAL="670688000" SDNUM="1033;1033;General"><FONT COLOR="#000000">670688000</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">USB1</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">USB-B-TH_67068-8000</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=LEFT VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">C114097</FONT></TD>
		</TR>
	</TBODY>
</TABLE>
[... cut ...]
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=CENTER VALIGN=MIDDLE SDNUM="1033;1033;General"><FONT COLOR="#000000">Bottom</FONT></TD>
			<TD STYLE="border-top: 1px solid #000000; border-bottom: 1px solid #000000; border-left: 1px solid #000000; border-right: 1px solid #000000" ALIGN=CENTER VALIGN=MIDDLE SDVAL="0" SDNUM="1033;1033;General"><FONT COLOR="#000000">0</FONT></TD>
		</TR>
	</TBODY>
</TABLE>
</BODY>

</HTML>

CSV files such as these must be generated from our KiCad BOM and Placement reports

ForrestErickson, 2022-11-30:
Lee has placed the three sets of files (Gerbers, BOM and Placement) into this folder:
https://github.com/PubInv/general-alarm-device/tree/main/Hardware/PolyVent_SPI_Controller/Rev0_1_Order20221129

RobertLRead, 2022-12-06:
This was ordered from JLCPCB, delivery is expected Monday, Dec. 12th.

Also, the Amazon long lead parts have been received.

closed 2022-12-06 by RobertLRead