PubInv/general-purpose-alarm-device #159: Fabricate Faceplate for /PolyVent_SPI_Controller for Robert and More

ForrestErickson, 2022-12-13:
@RobertLRead ,
@LokiMetaSmith 

Lee has created a stl file for a faceplate for the PolyVent SPI Controller board to anchor it into the Polyvent card cage.

[image]

See the file here: 
https://github.com/PubInv/general-alarm-device/blob/main/Hardware/PolyVent_SPI_Controller/CAD_model/PolyVent_SPI_Controller-faceplate%20v8.stl

ForrestErickson, 2022-12-13:
@RobertLRead 
@LokiMetaSmith 

Image with the printed circuit assembly in place.
[image]

RobertLRead, 2023-01-31:
This is a low priority but remains a valuable exercise.