PubInv/general-purpose-alarm-device #161: Trouble pulling CS low for SPI card....

RobertLRead, 2022-12-17:
I'm in the middle of debugging. This is for me to document my state of mind before I sleep.

I have the oscilloscope hooked up to the nCS test port. I believe the dip switches are correctly mapping the port i want, because I am controlling it in code, and can make changes in the time I attempt to pull it low which are reflected in changes correctly.

However, the pin I am using (CS6, which is back plane pin 13 as per: (spec)[https://gitlab.com/polyvent/polyvent_control_module/-/blob/master/wiki/2.1-Electrical-Specifications.md]) is never going fully low. On the scope it shows very brief drops to ground of about 10 nanoseconds. I'm trying to hold it low for 100 ms and I can't seem to do it.

Tomorrow, I will test this against backplane pin 13; that should answer if this is problem in my programming of the ESP32, or a problem in the card. I consider these equally likely at present.

ForrestErickson, 2022-12-17:
Commit the firmware so that I can try too.
CS6 is probably used on another card so try a CS that is not in the list as used.  FWWIS, I had difficulty finding the documentation on which CS lines were already in use and am not sure how fast I might find it again.

I am just off to a holiday party so it may be later or another day.

RobertLRead, 2022-12-17:
I found the documentation, your correct, that line is in use, I will change
it soon.

Here is the documentation:

https://gitlab.com/polyvent/polyvent_control_module/-/blob/master/wiki/2.1-Electrical-Specifications.md

ForrestErickson, 2022-12-21:
@RobertLRead 
I assume as of the success on 20221220 with the ESP32 in the PolyVent driving the GPAD this issue can be closed but will let you do so.

Lee

closed 2023-01-03 by LokiMetaSmith