PubInv/general-purpose-alarm-device #162: PCB Version 1, R103 and D102 (BUILTIN_LED) Incompatible With Level Shift on PolyVent_SPI_Controller Version 1

ForrestErickson, 2022-12-20:
# As built PolyVent_SPI_Controller Version 1 Not Driving GPAD Version 1 as built.
The PolyVent_SPI_Controller Version 1 as built in December 2022 was tested and not able to drive the SPI clock signal SCK to the full expected 5 volts when connected to the GPAD. 

## Problem Details
Screen shot at nCS (chip select), yellow, and SCK in blue. The SCK is measured at the interface to the GPAD and was expected to be 5V when high
[image]

## Problem Analysis 
Review of the design showed that the resistor and LED equivalent to the BUILTIN_LED of Arduino UNO was loading the level shifter. The level shifter circuit is a resistive pull up (not a totem pole logic output) adapted from an I2C level shifter. This resistive pull up scheme was too current limited to drive the signal to the required high logic level.

## Solution for GPAD Version 1.
Remove R103. 
20221220 Rob removed R103 and then the PolyVent was able to communicate to the GPAD.

ForrestErickson, 2022-12-20:
To do
Lee to change schematic for Version 2 PCB to make R103 DNI.
[image]

ForrestErickson, 2022-12-20:
Lee updated the rework notes at:
https://github.com/PubInv/general-alarm-device/blob/main/Hardware/Documentation/ManufacturingUnitTestTroubleshooting.md

to reflect "Remove and Relocate R103 and D102"  as per Version 2 PCB screen shot below.

[image]

RobertLRead, 2022-12-20:
This screen shot appears to highlight R112, not R102...can you clarify?

ForrestErickson, 2022-12-20:
Regarding " R112, not R102"

The rework is to  Relocate R103 and D102.
They become R112 and D106. 

I have created some what (I hope) better documentation in the document at:
https://github.com/PubInv/general-alarm-device/blob/main/Hardware/Documentation/ManufacturingUnitTestTroubleshooting.md

Which reminds me that firm ware should be changed to put the Wink LED onto A3 pin.
Which could be a Wokwi test too.

I will make a separate issue if there is not already one.

ForrestErickson, 2022-12-22:
Note direct drive of the GPAD with ESP32 3.3V signals is found to work (The GPAD received data and wrote date back to the ESP32). (It was found once the R103 is removed. but had Lee not put a 1K series termination in line with the ESP32 SCL line would probably have worked first try.

Signals using direct drive from ESP32 for nCS, SCK, COPI, CIPO for traces 1, 2, 3, 4.
For CIPO there is a 1K into 2K resistive divider providing 5V to 3V level shift and this measurement is at the divider.
[image]

LokiMetaSmith, 2022-12-27:
For version 2, relocate D106 such that it is visible with a shield mounted to the pcb.

closed 2023-01-31 by RobertLRead