PubInv/general-purpose-alarm-device #164: PolyVent_SPI_Controller PCB, Verion1 Limited Current For Pull Up, Add Logic Gate Buffer

ForrestErickson, 2022-12-22:
The level shifting method used in the PolyVent_SPI_Controller PCB is an "wired and" and relies on a pull up resistor for Voltage High on both the 3V side and the 5V side.  In version 1 as built by JCLPCB we specified 1K pull up resistors which we found out would not drive the SPI_SCK to a logic high when the UNO like on board BUILTIN_LED circuit was populated with a 330 Ohm and red LED.

Nathaniel had encountered the same problem when he had tried to interface to an UNO in the past.

Consider a change to all four SPI signal lines.
Adding SOT32 logic gate like such as:
* AND (onsemi, MFR.Part # NDS355AN, JLCPCB Part # C236917, cost 0.3255 at   qt = 10)
*  OR (Texas Instruments MFR.Part # SN74LVC1G32DBVR, JLCPCB Part # C10096  cost 0.09 at qt=10),
* EXOR (Texas Instruments MFR.Part # SN74AHC1G86DBVR, JLCPCB Part # C7471 cost $0.0656 at qt = 50 )

Would add only trivial cost but allow the drive of the SPI_Controller card to drive heaver loads.

The EXOR part could be configured by resistor selection to invert the signal if that were of any value for some TBD application other than SPI.

There is plenty of room on the PCB if we were to make a version 2.

ForrestErickson, 2022-12-27:
Here is a part designed to be a level shifter:

https://www.ti.com/lit/ds/symlink/txu0204-q1.pdf?HQS=dis-mous-null-mousermode-dsf-pf-null-wwe&ts=1672160292672&ref_url=https%253A%252F%252Fwww.mouser.com%252F

LokiMetaSmith, 2022-12-27:
A proper Translation - Voltage Levels Four channel unidirectional level shifter should be identified