PubInv/general-purpose-alarm-device #170: Team Review of Firmware/GPAD_API/README.md

ForrestErickson, 2022-12-26:
Some time ago I wrote a readme which to the best of my knowledge described the GPAD_API firmware at that time.
It must contain some misunderstandings on my (Lee) part.

I know it is not complete. (It does not cover the description of possible buzzer sounds and amplitudes for example. It does not utilized the PWM that is available on some but not all of of the LIGHT channels.  It does not utilized all of the features of the LCD display.)

Document to which I referred is at:
https://github.com/PubInv/general-alarm-device/blob/main/Firmware/GPAD_API/README.md

LokiMetaSmith, 2022-12-27:
API needs to include the SPI interface as well as a command list from the serial interface.

RobertLRead, 2023-01-25:
I have included the SPI interface in the documentation here, as well as referencing and giving more documentation in the GPAD_API_SPI_controller, as well as cleaning up the code there a bit.

closed 2023-01-25 by RobertLRead