PubInv/general-purpose-alarm-device #177: Firmware Testing, Branch: Rob-Create-Serial-Transfer-SPI-Pair, GPAD_API_SPI_CONTROLLER.ino and GPAD_API.ino

ForrestErickson, 2023-01-12:
# Testing Summary
On 20230111 Lee used **git pull** and updated his local clone of the PubInv repository.

* The Controller side has no LCD display but shows serial port reporting of data being sent to the SPI interface.
* I do not get any alarm results on the Peripheral GPAD side on the serial port nor LCD nor LIGHTS nor Buzzer. 
* I think I have found corruption of the GPAD such that loading even the I2C scanner does not work
**I recommend no other GPAD devices be loaded with the Firmware in the, Branch: Rob-Create-Serial-Transfer-SPI-Pair**

#Testing Notes
## Journal of "Out Of The Box" testing
Switched to the Rob-Create-Serial-Transfer-SPI-Pair branch.
Loaded firm ware GPAD_API_SPI_CONTROLLER.ino into GPAD SN#6(COM16) and and GPAD_API.ino into GPAD SN#2(COM15).

Neither device has LCD display.
On the controller GPAD_API_SPI_CONTROLLER.ino into GPAD S#6(COM16) we see the following serial port activity.
[image]

On the GPAD S#2(COM15) the serial port does not show program name but only what I assume is a version number " 0.07"
[image]
There is no further activity.

## Other testing
I am concerned that the LCD is not displaying.

On the Controller
On GPAD S#6(COM16)  loaded the I2C scanner program and the expected message of "I2C Scanner" only appears as " I2C"
The address for the I2C interface for the LCD does NOT appear.
[image]

On the Perihperial
On GPAD S#2(COM15) loaded the I2C scanner program.  The expected message of "I2C Scanner" appears as well as "Scanning..."
The address for the I2C interface for the LCD does NOT appear.
[image]

# Caution
This is very concerning.  The "\nI2C Scanner" message is part of the setup() function of the I2C scan and the message "Scanning..." is in the loop().  The two GPADs are not operating normally and not even operating the same.

**I recommend no other GPAD devices be loaded with the Firmware in the, Branch: Rob-Create-Serial-Transfer-SPI-Pair**

ForrestErickson, 2023-01-13:
20230113 Resume testing.
Tried to reboot the Arduino IDE.  It failed to close. Reboot PC.
Starting fresh. Lets load the "i2c_scanner" into each DUT.

* GPAD S#6(COM16)  
This Device had the GPAD Controller firmware once loaded. THen I loaded the Factory Test V1.30 to try to rest it.
This screen capture shows the Factory test version 1.30 failing to find the I2C LCD interface and is followed by the I2C scan which does not find the LCD interface nor does it terminate as it should.  
[image]

* GPAD S#2(COM15)
This screen capture shows the version 0.07 of the Peripheral device GPAD_API followed by the I2C scan which does not find the LCD interface nor does it terminate as it should.  
[image]

Note that the two GPADs fail the I2C test differently.

Here is oscilliscope captures on both I2C busses. 
[image]
[image]

* CH1 is GPAD S#2(COM15) SCL 
* CH2 is GPAD S#2(COM15) SDA
* CH3 is GPAD S#6(COM16) SCL 
* CH3 is GPAD S#6(COM16) SDA

Note that the SCL is not 5Vp-p as it should be.

RobertLRead, 2023-01-13:
This is very confusing; do we believe that SOFTWARE caused this error?  It
seems to have occurred upon loading my new software, but it is hard to
understand how it could permanently damage the LCD even when other software
is reloaded. Is that what we are observing?

ForrestErickson, 2023-01-13:
Measure SDA at the I2C expander chip U301 and there it is also of reduced amplitude. The pull up resistor on SDA R108 is going to +5V (same as the pull up for SCL).  So either the I2C expander chip U301 or the ATmega328 U102 is loading the signal. 

Here CH1 is SCL at the header pin and CH2 is SDA right on U102 Pin 27.
[image]

Next I hold U102 in reset. 
[image]
SCL goes to +5 but SDA only goes to 2V.

Need to find what is clamping SDA to no higher than 2Volts.

ForrestErickson, 2023-01-13:
Regarding, "do we believe that SOFTWARE caused this error?  "
It is hard to image a way that it could.  That would be some mighty powerful voodoo to do that.
The I2C line is supposed to be a "wired or" connection where a resistor pulls the line high (5V) and either the controller or peripheral can sink current to pull low.  The only way their might be a chance to make something bad happen would be if the Peripheral tried to pull the SDA low while an incorrectly configured Controller SDA had an active pull up to high. Then their would be high currents in both parts.  A failure on IO pins would usualy be to GND or to VCC. It is hard to imagine to 2V.

Back to pondering.
On the SDA line there are four connections. 
1. U102 the uController, 
2. U301 the I2C expander, 
3. the pull up resistor R108 and 
4. the series resistor R105 which brings the signal out to J105.

If  it were easy to unsolder either or both integrated circuits that would leave only the pull up resistor to +5 and I would bet a shinny copper penny the line would go high to +5. But it is not easy to unsolder either part. 

Next I try a second 1K pull up in parallel to R108 to see if the SDA goes higher.  I connect at the header pin for A4.
Measure the voltage and only goes from 1.85 to 3.14 as measured by multimeter.  ALSO the SCL and SDA waveforms stop having square waves and go to high. SCL to 4.8 and SDA to 2.2 as measured by oscilloscope. 
[image]

Pulling SDA to GND through the 1K and the voltage goes to 1.65.
[image]

Next Get Insight Into Where Current on SDA is To and From.
Measure the current into the uC U102 by measuring the voltage across R101. This is done by measuring the voltage at +5V test point minus VccController test point.  When the system under test is in reset the voltage is 6.8 mV (current draw is 6.8 mA) and when I load SDA line with additional 1K the voltage becomes 7.2mV (current draw is 7.2mA) so there is a difference of 0.4 mA.
For a 1K resistor on about 2.2V to GND the current is about 2.2 mA.  
The VccController net is supplying the pull up for R108 so the change of 0.4 mV at VccController is in part do to the change in voltage as SDA has 2.2 mA pulled from through R108.

Measure the current into the uC U301 by measuring the voltage across R310. This is done by measuring the voltage at +5V test point minus VccLCD test point.  When the system under test is in reset the voltage is 59.5 mV (current draw is 59.5 mA) and is the same when the additional 1K to SDA is added to GND. Therefor it appears that the current is NOT going through U301 the I2C expander.

## Insight from SDA loading with 1K.
Lets restate. With the I2C scan firmware running there are square waves on the SCL and SDA. As soon as I load the SDA to GND with an additional IK the square waves stop.
If I load SDA with 1K and press reset the DUT never has square waves on SCL and SDA.
If I load SDA with 1K to VccController and press reset the DUT never has square waves on SCL and SDA.

ForrestErickson, 2023-01-13:
20230113 14:07 Resume testing. My Arduino IDE locked up and the RealTerminal I had connected to GPAD S#2(COM15) also would not exit.  

I tried terminating both in the Task Manager with success on the Arduino. But no success on the RealTerminal.
I rebooted the computer.

Preform the following experiment. 
Connect to GPAD S#2(COM15 with RealTerm. Press reset.  the I2C scan does not terminate nor find the LCD.

Connect additional 1K from SDA to +5V. Press reset.
The I2C scan finds the LCD address.
[image]

How unexpected.

Lets Test an Arduino UNO with the I2C Scan.
[image]
About every 5 seconds the scan reports "No I2C devices found"

Now add 1K pull down resistor on the SDA and reset.
[image]
The scan gets stuck just like GPAD was!

**Tentative conclusion.** 
There is a current path clamping the SDA low enough that the uController goes not get valid communication on the I2C scan. 

## Multimeter Tests
Remove power and USB connections to UNO and GPADs.

1. On Arduino UNO RAW with out pull up on SDA with multi meter I measure 1.7 Meg from SDA to GND and 846K from GND to SDA
2. On Arduino UNO with 1K pull up to +5V multi meter I measure 2.45K from SDA to GND and 2.45K from GND to SDA.  When I reverse the meter connections making the GND to SDA measurement there is some capacitance charging and it takes some 5 seconds or so for the measurement to reach it's final value.

3. On GPAD S#2(COM15) with multi meter I measure 2.00k from SDA to GND and 2.00K from GND to SDA
4. On GPAD S#6(COM16) with multi meter I measure 2.00k from SDA to GND and 2.00K from GND to SDA

ForrestErickson, 2023-01-13:
On GPAD S#2(COM15) I connected a 1K resistor from SDA to +5V. 
I have checked out the repository at TAG V1.2
I lave loaded the Factory Test firmware into GPAD S#2(COM15). This is Factory Test Version V1.24.

The device boots up and the LCD shows the message.

@RobertLRead 
Please lets see if we can recover LCD function on your GPAD(s) with the V1.2 **Factory Test firmware V1.24**
You will need to connect a 1K resistor from the SDA net to +5V. 
I will make additional information in a later post. 

Once again I have to reboot my computer because the Arduino IDE has locked up.

ForrestErickson, 2023-01-13:
How I switched to the Tag with the Factory Test V1.24
[image]

ForrestErickson, 2023-01-13:
The 1K resistor plugged into A4 socket and clip lead to +5V.
[image]

ForrestErickson, 2023-01-13:
@RobertLRead 
@LokiMetaSmith 

Regarding the trail of trouble shooting notes above, I conclude that something is sinking current from the SDA line on both of my GPAD units S#2 and S#6. It was noticed when I tried to evaluate Rob's branch "Rob-Create-Serial-Transfer-SPI-Pair"
I think it happened when I loaded and ran this firmware for the Controller and the Peripheral.   

**HOWEVER**, I can also say that it has been some time since I was running any firmware which addressed the LCD because I had been working with the buzzer for quite some time and none of that firmware had LCD code in it.

So far as I can figure out, with out cutting a trace, I cannot further isolate U201 from U301 as the path for the current sink.
I am going to let this stew in my brain for a bit before I cut traces. Regarding cutting traces I will not be very hard but I want to stew a bit before I cut.

I have a clear access to the trace where I could cut as highlighted in this screen shot.  
[image]
It would be easy to repair after the cut.

Once cut and power applied to the DUT the SDA will either:
1.  go up to 5V which would indicate that the uController U201 is at fault, 
2. Or SDA will go up to about 2V indicating the I2C expander is at fault 
3. or something else indicating I have not figured out all the contingencies.

Regardless I am going to stew.

In order

RobertLRead, 2023-01-13:
I can endeavor to do this tonight when I have instructions. My rework
ability may not be as developed as yours is.

ForrestErickson, 2023-01-14:
I used at 1K pull up resistor and that is a sample size of one.   Assuming the same problem, the leakage may still be different from assembly to assembly. 

On both S#2 and S#6 I have SDA at 2.64 Volts. Holding the uController in reset by holding S101 closed should make the measurement independent of any firmware.

ForrestErickson, 2023-01-14:
Simpler test would be to straddle a leaded resistor to the very next pin for +5V at this indicated spot.
[image]

Soldering a resistor on top of R108 would also work.
[image]

RobertLRead, 2023-01-14:
Dear Lee,

I have attempted to perfrom this experiment on board #5. I have clipped a 5V signal off the left of the card with an aligator clip to a 1168 Ohm resistor (measured with my multimeter.) This measures a steady 5.035 V against a ground measured from one of 4 mount holes on the board. Board number 5 does not have a header soldered in, but I place it in the whole and bend it so that I am pretty sure it is making contact. When I reset the card, the LCD still does not activate. I have followed your instructions have reverted to code GPAD_Factory_Test v1.24. The output of the serial port at reboot is shown below:
[image]
[image]

ForrestErickson, 2023-01-14:
@RobertLRead 
That looks like it should work.
With a multi meter set for DC voltage measure at the resistor lead connecting to the PCB as indicated here. 
[image]

RobertLRead, 2023-01-14:
5.27 Volts

ForrestErickson, 2023-01-16:
20230116 7AM
Because Robert found a different voltage I suspected some difference and found it. 
I introduced an error on my two assemblies when I relocated the BUILTIN_LED off of D13 to A3 to become a WINK_LED (and resistor). 
I mistakenly moved the relocated LED to the SDA trace not A3.  Once I moved the LED and resistor the SDA had normal amplitude and the static measurement of SDA is 5.02 VDC.
I will load an updated photo of the rework.

ForrestErickson, 2023-01-16:
Rework

closed 2023-01-16 by ForrestErickson

ForrestErickson, 2023-01-16:
The problem @RobertLRead  is having with LCD on a unit remains unresolved.

ForrestErickson, 2023-01-16:
I have annotated the SCL and SDA pins on Roberts photo.
[image]

Here are hilighted traces for SCL up to the pin on U301 Pin 14 too.
[image]

Same for SDA  U301 Pin 15
[image]

And the power into U301 Pin 16
[image]

Ground is on pin 8, the diagonal corner from power.