PubInv/general-purpose-alarm-device #194: Develop a circuit that can drive a piezo to fire alarm audio levels

LokiMetaSmith, 2023-01-31:
The existing driver doesn't sufficiently drive the piezo and it isn't obvious why. 

This requires more research.

ForrestErickson, 2023-02-16:
Notes on Piezo Buzzers.
Calculation for inductor for resonance at 4KHz
L=1/(4*Pi^2*F^2*C)

| Digikey Part Number | Description               | Capacitance | Inductor For Resonance | Note  |
|---------------------|---------------------------|-------------|------------------------|---|
| 490-4698-1-ND       | Buzzer Piezo 1.vV 12.6mm  | 5.1nF       | 0.317H                 |   |
| 2104-TP134005-1-ND  | BUZZER PIEZO 5V 13.8MM TH | 11nF        | 0.143H                 |   |
| 445-2526-1-ND       | BUZZER PIEZO 5V 14MM TH   | 12.3nF      | 0.129H                 |   |
| 433-PT-1407T-ND     | BUZZER PIEZO 5V 13.8MM TH | 10 to 15 nF | 0.158 to 0.106H        |   |
|                     |                           |             |                        |   |

I suspect these inductors will be gigantic.

ForrestErickson, 2023-02-16:
A search on Digi-Key for  inductors from 100 to 330 mH produced 29 hits.
When sorted from lowest price they start at $0.78
https://www.digikey.com/en/products/filter/fixed-inductors/71?s=N4IgjCBcoEwAwA4DsVQGMoDMCGAbAzgKYA0IA9lANogDM8ABALYASIpYccTr7DLb4AKxd%2BpeCJ4gYSCQJqduIALqkADgBcoIAKoA7AJbqA8pgCyhbPgCuAJ0IgAvqQC0MVCAyR1NqyXJUQQQEATmUnEGcad09vX1IKSGoOMPCANnd9ABMtZw4INU1IEAF1AE9VeyLLDAcHIA

[image]

ForrestErickson, 2023-02-16:
The lowest price inductor is RLB0913-104K
https://www.digikey.com/en/products/detail/bourns-inc/RLB0913-104K/6677257
This has a series resistance of 235 Ohm,
The Q will be XL/R which from this on line calculator  
https://www.everythingrf.com/rf-calculators/inductor-quality-factor-calculator
is predicted to be very UNBELEABABLEY high at 10689361. 

Too good to be true.

ForrestErickson, 2023-02-16:
Found another on line calculator at: https://eliterfllc.com/rf-calculators/inductor-quality-factor-calculator

For L=100mH, R-235 and f=4,000 it produced a much more believable Q of about 10.
> [image]

For a Q of 10 I think the bandwidth will be about 1 in 10 and therefor the sensitivity to the center of self resonance will be something like +/- 400 Hz ( or f/Q).

ForrestErickson, 2023-02-16:
This inductor would be approximately correct value for the three largest transducers which have higher capacitance.  

The inductor resistance of 235 Ohms limits current to 5V/235Ohm or about 21 mA which is much less than the moving coil current of approximately 5V/8Ohms or 623mA (not taking reactance into account).

### Tentative Conclusion
With out testing, I am not sure I can be comfortable with a design with this approach.

LokiMetaSmith, 2023-02-16:
https://tinyurl.com/2jmoyov9

I think the inductor selection is primarily tied to the selection of the
max voltage. too large of an inductor can create a waveform that can damage
the piezo element though

LokiMetaSmith, 2023-02-16:
https://www.mouser.com/ProductDetail/Taiyo-Yuden/NRS5020T3R3NMGJ?qs=PzICbMaShUdfk%252BLDXl%2FXZw%3D%3D

This one looks correct to me and is size appropriate.

LokiMetaSmith, 2023-02-16:
ok, also note that the one I posted with a sample circuit, the data sheet
for the buzzer has a very important line. " Internal Circuit-
Audio-oscillator and piezoelectric driver"
It's got an internal Audio Oscillator and piezoelectric driver, which is
why the circuit is a non-excitation circuit.

The circuit simulator example I posted is based off of a piezoelectric
drive circuit and generic piezo element, and should be easy to test with
existing boards.

I agree that using a 200ohm inductor won't work, but it looks like in
simulation that a 2.7-3.3uH inductor will, and they come in low ohm
resistive varieties.

ForrestErickson, 2023-02-16:
@LokiMetaSmith 
Regarding, "I agree that using a 200ohm inductor won't work, but it looks like in simulation that a 2.7-3.3uH inductor will, and they come in low ohm resistive varieties."

Please show simulation.
By what criteria will you determine "work"?

Maximum energy transfer will be at resonance For a 3.3uH inductor at 4KHz the reactance is about 82 Ohms 
[image]
At 82 ohms, the current will be on the order of 5/82 or 60mA.  Power would be on the order of 4.3^2/82= 0.22W.
Far less than the current speaker with moving coil at 4.3^2/8= 2.3W.

LokiMetaSmith, 2023-02-16:
[image: image.png]
https://tinyurl.com/2jmoyov9

Lee,

I think you need to check your equations.

https://www.electronicdesign.com/power-management/article/21808584/use-resonance-with-a-highvoltage-piezo-driver
I did a rough calculation with a 10nF piezo and got 9uH as the inductor at
4kHz

ForrestErickson, 2023-02-16:
Ok thanks for the link to the simulation.
I do not believe simulation is telling you that this is "working" at all.

The maximum audio output of the transducers is on the order of 4KHz. With the buzzer horn on the unit in my lab the maximum output per my ears moves about 4K to a lower frequency of about 2.4 KHz I assume because of tighter coupling to air loads the peizo.

Your simulation is with a square source at 2Khz and the LC response to the step is the damped oscillation. The repetition rate of the oscillation is .25mS which I am going to guess is 1/2*(1/2KHz) . The damped oscillations are orders of magnitude higher in frequency. I think I count about ten peaks in 24uS which is  about 2.4 uS per peak or about 40KHz.

In this screen shot the burst starts just at 17.00mS and about 17.024 is where I think I have counted about ten peaks. If the time resolution could be increased on the simulation this would be easier to estimate.
[image]

**Bottom line**
The piezo is driven way above the audio range and little or nothing to hear.

ForrestErickson, 2023-02-16:
I set the simulation for sine wave source and got this result.
Screen shot.
[image]

From this we can see clearly what 2Khz (the sine wave) looks like and that the resonance response is way higher in frequency per my analysis above.

ForrestErickson, 2023-02-17:
I am not sure if I am changing your simulation when I do what I have done so hope it does not interfere with your intentions.

I have found how to zoom out in time and it appears to be zoomed in the maximum at this view which is no better than my previous screen shots.  However I did turn on the waveform for the source and messed with the values to make it go from zero to 5V.

URL for the simulation as I have changed it:
https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCDsB0AcYCYCsBTAtJEAWaAGSsSkkAzEkgJwnUmwhK732OpphgBQALuAGyMIEWEMl4ihIRhDZwKFXiTkJI+SkliZc0SEjCxlvSApLsqScLkYATFADMAhgFcANlw4BjENTFgKCLyRiQoyMOCqWEZGRWOh0CNC8umD4WOQUsPLE-lqWnADuIkhBuHFF4EaSHAXe5WI1vtlVAUFYjDWildWBhUFlHbhNHUN9rZXOPeAIpWK8wiF4kcogaFpYQmBkJJZkVGCQFJUA5hMkCAJlgXQhHABKE7Mn2ZLYbc-zSLcg8uLCWHESAhEESYIUk0A+ACcvkEJGB+D9nslLBwocNenUzm9oJxUWVENNalicRM9nVuqSiRwAPbgWlSQQUULaYgpZLqBi8CiaBa4CCAiAkDhAA

ForrestErickson, 2023-02-17:
I have learned to make a copy of the circuit. I have saved it locally.
I have also made a duplicate but changed the inductor to 330mH and resimulated.  
Screen shot
> [image]

Note: This simulation does not reflect the resistance of a real inductor.

**Bottom Line**
Here we see the voltage driving the piezo is now following the source frequency and is reaching nearly 9 V peak.
The very large inductance is needed to give the hope for boost

ForrestErickson, 2023-02-17:
Another circuit added to the simulation with a diode to allow the resonance to swing negative.

I also chose L so that the resonant frequency is better fit to the driver frequency which gives more driving time (less dead time)
I set the Vcc to +5. 

[image]

The resulting waveform swings to about +15 and -5 for about 20Vp-p.
I would have to assume this is about 20Log(25/4.3) louder or 14 dB more.  Note the specifications for the Piezos say the response is lower for sinusoidal drive than square wave so this is not an apples to apples comparison.

LokiMetaSmith, 2023-02-17:
Lee, you can share a link that will share your modifications.

A few things, unless the piezo can run at 20v p-p that would destroy our
current selection of buzzers, also I'm not sure you want the speaker at
resonance, This driver circuit gives a chirp every level transition on the
rising or falling edge of the drive PWM
[image: image.png]

ForrestErickson, 2023-02-17:
Does this link do it: 
https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCDsB0AcYCYCsBTAtJEAWaAGSsSkkAzEkgJwnUmwhK732OpphgBQALuAGyMIEWEMl4ihIRhDZwKFXiTkJI+SkliZc0SEjCxlvSApLsqScLkYATFADMAhgFcANlw4BjENTFgKCLyRiQoyMOCqWEZGRWOh0CNC8umD4WOQUsPLE-lqWnADuIkhBuHFF4EaSHAXe5WI1vtlVAUFYjDWildWBhUFlHbhNHUN9rZXOPeAIpWK8wiF4kcogaFpYQmBkJJZkVGCQFJUA5hMkCAJlgXQhHABKE7Mn2ZLYbc-zSLcg8uLCWHESAhEESYIUk0A+ACcvkEJGB+D9nslLBwocNenUzm9oJxUWVENNalicRM9nVuqSiRwAPbgWlSQQUULaYgpZLqBi8CiaBa4CCAiAkDhAA

ForrestErickson, 2023-02-17:
Trying again: 
https://tinyurl.com/2qzp53ky

ForrestErickson, 2023-02-17:
Regarding, "...A few things, unless the piezo can run at 20v p-p ..."
Here is the detail from the JLCPCB part in the Version 1.
[image]

ForrestErickson, 2023-02-17:
Notes on candidate inductor part: 
https://www.mouser.com/ProductDetail/Fastron/11P-154K-50?qs=sGAEpiMZZMv126LJFLh8yz21YmpXffL45iYhXVG3qrw%3D

ForrestErickson, 2023-02-17:
Best candidate from DigiKey is 120mH
https://www.digikey.com/en/products/detail/bourns-inc/RL181S-124J-RC/946962

ForrestErickson, 2023-02-17:
@LokiMetaSmith ,

Capturing email traffic regarding this issue
[image]

Looking up the IRLML6401 I see that it is a P channel Enhancement mode MOSFET.
While I am fluent with BJTs MOSFETs are almost tera incogneta for me.
From what little expernce I have with such circuits the N Channel Enhancement mode MOSFET can be used much like a an NPN BJT as a switch where the emitter is common to ground and through a base resistor the transidter is turned on to saturate.
An N Channel Enhancement mode MOSFET can be used similarly with the source grounded and the device turns on when the gate is more positive than the threshold voltage.

This P Channel appears to me to be operating as a source follower not a switch which seam like it might work but with a threshold voltage drop across the Source to Drain dissipating more power than if a switch were used.

**Have I assessed and described the operation of the  IRLML6401 correctly?**

I am grouping towards what combination of circuits to put into the schematic still.

LokiMetaSmith, 2023-02-17:
Lee,

The buzzer circuit there uses a piezo with an integrated excitation
circuit. It's part of why it costs more. The mosfet is acting like a low
side switch to emit a constant high volume beep.

I'm not sure exactly why the design used a p-fet, I don't think the
application matters as it' only being used to turn the buzzer on or off.

ForrestErickson, 2023-02-17:
@LokiMetaSmith 
Regarding, 
"The mosfet is acting like a low side switch to emit a constant high volume beep."  I suspect it is more like a source follower than a switch. But just like an emitter follower could be used to turn on the buzzer it works.  

In other news....
I have pushed successive simulations to the repository at:
...\general-alarm-device\Hardware\GeneralPurposeAlarmDevicePCB\Simulations
There is a README.md there with some notes.

Unfortunately, I have no inductors in the 100mH range with which to experiment. I am not of steam with respect to trying to fabricate one at this time of day also.

LokiMetaSmith, 2023-02-17:
Me neither. we might need to admit that the buzzer will not be loud enough
for alert notifications or fulfill it's intended purpose.

In other news, I'm concerned that we have not done sufficient research
regarding standard FDA medical device guidelines for life saving medical
devices.
https://www.fda.gov/medical-devices/device-advice-comprehensive-regulatory-assistance/human-factors-and-medical-devices

It is out of the scope of this project, but clearer guidelines would be
invaluable in avoiding re-inventing a slightly less optimized wheel.

ForrestErickson, 2023-02-17:
@LokiMetaSmith 

Regarding, "we might need to admit that the buzzer will not be loud enough for alert notifications or fulfill it's intended purpose"

Robert has ordered for me the very loud fixed tone devices and they will arrive next week.

ForrestErickson, 2023-02-27:
Video demonstrating the piezo with driver circuit.
https://www.youtube.com/watch?v=tq2dxpzMS78  See at 6:30

The AdaFruit Buzzer has mechanical dimensions: 

- Dimensions (not including pins): 12mm diameter, 9.7mm tall.
- Pins are 6mm long and 0.3" apart.

The diameter is very similar to the part we put into Version 1 PCB from JLCPCB HOWEVER the pin spacing is greater than the 200 mills of the Version 1 part.

ForrestErickson, 2023-02-28:
Per discussion of 20230228 Lee and Lawrence we will go forward with the P1536A on the Version 2 PCB.