PubInv/general-purpose-alarm-device #196: Hardware Failure Report, GPAD Version 1, Serial #1, Intermittent LCD

NULLbodyHERE, 2023-02-04:
**Describe the Failure / Bug**
A clear and concise description of what the failure / bug is.
Short Summary of the expected behavior: LCD display should show the default booting message and serial monitor.
Short Summary of the buggy behavior: Intermittent displaying of bad/invalid characters (as shown in the picture)
[image]

**Device Under Test, aka DUT**
Steps to reproduce the behavior:  
1. Model Number:  GPAD Version 1
2. Serial Number(s): 1
3. Location where failure / bug occurred: Benign environment after international shipping.
4. Hardware configuration. (How connected, any accessories, what else it takes to replicate the system in which the failure occurred.): default configuration (no accessories)
5. Extent of Failure: intermittent

**To Reproduce**
Problem is not yet reproducible because it is intermittent.

**Expected behavior**
Upon booting, the screen should show the default booting message. After booting, the display should allow text to be written to via the serial monitor.

**Hardware description
Model: GPAD Version 1
Serial Number: 1

**Test Equipment
Power: wall transformer 12V
Data Line: Micro USB

NULLbodyHERE, 2023-02-04:
# Testing Ideas

- With a multimeter, test continuity on every pin from U301 to a trace.

- While tapping on the DUT (Device Under Test), press the reset button and observe the boot message on the serial monitor.

ForrestErickson, 2023-02-04:
Lee suggests take the factory test firmware any version.
Copy to a Troubleshooting or folder TBD with in the Factory Test folder.
Name it descriptively like "U301 Dynamic Test"

Then take the LCD splash screen function and wratp it in a while(1) loop
In the loop write to the display a series of ASCII characters which end in 0x?5 and 0x?C where ? is chosed for something alpha numeric.

This should cause ever data pin from U301 to the LCD to be very active.

ForrestErickson, 2023-02-06:
@NULLbodyHERE 
FYI I have created an elementary slow speed voltage plotter (like an oscilloscope) for an Arduino UNO here:
https://github.com/ForrestErickson/SerialPloterA0A1A2A3A4A5/blob/main/README.md

NULLbodyHERE, 2023-02-07:
@ForrestErickson,
Thanks very much for this. I'll take a look into it. Much appreciated.

ForrestErickson, 2023-02-08:
@NULLbodyHERE 
I would appreciate if you make note of anything that is confusing so that I can improve the instructions.

With in the code I have notes about how to change to graph legend and sample rate. These changes happen only at compile time.

It was interesting for me to force failures on the I2C buss and learn how the firmware would or would not react.
Once you get Serial Number 1 free of intermittent this could be fun to investigate further. 

We might also investigate simulations with Wokwi for failures too.