PubInv/general-purpose-alarm-device #199: Using Aferro GPL preamble on all our code...

RobertLRead, 2023-02-07:
Dear @ForrestErickson,

Although your license:

`/* SPI_CONTROLLER
   From: https://circuitdigest.com/microcontroller-projects/arduino-spi-communication-tutorial
   Modified by Forrest Lee Erickson 20220523
   Change to Controller/Peripheral termonology
   Change variable names for start with lowercase. Constants to uper case.
   Controller Arduino Code:
   License: Dedicated to the Public Domain
   Warrenty: This program is designed to kill and render the earth uninhabitable,
   however it is not guaranteed to do so.
   20220524 Get working with the SPI_PERIPHERAL sketch.
   20220525 The SCK period is about 250nS as set up.
   20220927 Change for button to have internal pull up.  Add Resistor and LED from pins 7 to 6.
*/`

... is humorous, I would prefer that we stick to the policy of using the Affero GPL license as exemplified here:

`

 /* GPAD_API.ino
  The program implements the main API of the General Purpose Alarm Device.
  Copyright (C) 2022 Robert Read
  This program includes free software: you can redistribute it and/or modify
  it under the terms of the GNU Affero General Public License as
  published by the Free Software Foundation, either version 3 of the
  License, or (at your option) any later version.
  See the GNU Affero General Public License for more details.
  You should have received a copy of the GNU Affero General Public License
  along with this program.  If not, see <https://www.gnu.org/licenses/>.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
  Change notes:
  20221024 Creation by Rob.
  20221028 Report Program Name in splash screen. Lee Erickson
*/` 

Although I often forget, and don't want to cramp your style, and important best practice for any open-source project is to put such a preamble on every code file.

ForrestErickson, 2023-02-08:
@RobertLRead 
I had not realized my license had made it into any of our code and if I had been asked I would have said I had looked. 

I think I have been copying into any code I put on Public Invention the license from the GPAD_API so I guess some review necessary.

I just now checked the toneSweep and see that there is not a license of any sort in the code and I remember it is because I made it by modifying an Arduino IDE example from which I think there is no license.
I think the example from which it stems was from the tone library

What is the appropriate thing to do?

ForrestErickson, 2023-02-08:
A bit of pondering and I figured out how to find the code from which I made the toneSweep. It was the toneMelody example found here:  https://www.arduino.cc/en/Tutorial/BuiltInExamples/toneMelody
I see I removed the public domain part of it but did not put in the Public Invention license.

Unless there is a reason not to do so I will do so.

RobertLRead, 2023-02-08:
You want the first comment to look like the example below (this is taken from the bottom (end)) of the GNU Affero explanation:

[https://www.gnu.org/licenses/agpl-3.0.en.html#:~:text=The%20GNU%20General%20Public%20License,becomes%20available%20to%20the%20community.](https://www.gnu.org/licenses/agpl-3.0.en.html#:~:text=The%20GNU%20General%20Public%20License,becomes%20available%20to%20the%20community.)

`/* 
alarm_api.cpp
  This implements the "abstract" alarm api.

  Copyright (C) 2022 Robert Read
  This program includes free software: you can redistribute it and/or modif
  it under the terms of the GNU Affero General Public License as
  published by the Free Software Foundation, either version 3 of the
  License, or (at your option) any later version.
  See the GNU Affero General Public License for more details.
  You should have received a copy of the GNU Affero General Public License
  along with this program.  If not, see <https://www.gnu.org/licenses/>.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
*/`

ForrestErickson, 2023-02-09:
Notes:
I made a pull on the local clone of the repository to get updated.
I opened the files and I copied the above license and replaced to old license.  I saved.
I added the files to commit locally and got the following message;

[image]

So I reset to un-stage the commit.
[image]

I am guessing the .git folder in the is left over from when this was on my personal repository so I am going to delete it.
[image]

Now try again to stage the files.
[image]

I think this is going to work.

I have pushed the changes.
[image]

RobertLRead, 2023-06-20:
Rob will check now.

RobertLRead, 2023-06-20:
done

closed 2023-06-20 by RobertLRead