PubInv/general-purpose-alarm-device #203: Finalize order and purchase new PCB (Rev 2)

RobertLRead, 2023-02-14:
Review the Rev 2 PCB, then order 15 new populated boards.

Also, order 5 3d printed cases.

Our goal here is to have at least 3 fully functional devices to have at the OSHWA table in April 28th in New York.

LokiMetaSmith, 2023-02-14:
- [ ] Make modifications for louder Piezo
- [ ] Review GPAD schematic
- [ ] Review 3D printed STL files
- [ ] Review Design for Manufacturing JLCPCB
- [ ] Review gerber files
- [ ] Export gerber files
- [ ] Export STL files
- [ ] Order

ForrestErickson, 2023-02-21:
On 20230221 Lee has fixed the pin numbering problem on the I2C connector J105 and moved J105 closer to the edge of the PCB.
As of 10 AM still reviewing issues for more actionable items.

ForrestErickson, 2023-02-21:
Updating some silk screen locations. Run DRC and get only the expected warnings.
[image]

ForrestErickson, 2023-02-21:
Generated step file of the PCB produced  a large report with a lot of parts not converting.  Not a blocker but something for information only.

RobertLRead, 2023-06-20:
Done.

closed 2023-06-20 by RobertLRead