PubInv/general-purpose-alarm-device #207: Fix Gap Between Features on

ForrestErickson, 2023-03-09:
Here is a gap that was in file:
BackPanel_Hanging-Compound.stl
[image]

Fixed it with file:
BackPanel_Hanging-CompoundV2.stl
[image]

ForrestErickson, 2023-03-09:
Gap as seen on the GitHub preview file: BackPanel_Hanging-Compound.stl

[image]

ForrestErickson, 2023-03-09:
No gap seen on file BackPanel_Hanging-CompoundV2.stl

[image]

ForrestErickson, 2023-03-09:
Chat history with JLCPCB on 20230309 about 4:10 AM

> 

 	
Name(Account #): 2999181A

What do you need help with? My Orders

 
 
 
 	Sinclairzhu	Thu, 03/09/23 04:12:46 am America/New_York	
 
Welcome back, may I help you?
 
 
 
 
 	2999181A	04:12:46 am	
 
Hello
 
 
 
 
 	Sinclairzhu	04:12:58 am	
 
Hi sir
Good day
 
 
 
 
 	2999181A	04:13:00 am	
 
Hello,
I finally understood the problem reported with my order on part D2023030341500102-29
 
 
 
 
 	Sinclairzhu	04:14:54 am	
 
let me check please
 
 
 
 
 	2999181A	04:15:17 am	
 
I have uploaded a new file with a new name indicating V2 for Version 2.
[image.png](https://cdn.livechat-files.com/api/file/lc/att/12327960/74f993b2c64d92fb0fc3e56bbcff26e4/image.png)
I have sent an email back to the support email address also describing the changes.

Is their anything else I might do to make sure JLCPCB had everything they need to proceede?
 
 
 
 
 	Sinclairzhu	04:17:19 am	
 
I have pushed our engineer to reply you ASAP sir
 
 
 
 
 	2999181A	04:17:40 am	
 
I am at 4:17 AM which is UTC-5 hours. What is your time there?
 
 
 
 
 	Sinclairzhu	04:18:15 am	
 
we are 5.18pm now
 
 
 
 
 	2999181A	04:19:41 am	
 
Ok thanks so that probably means you are at UTC +8.
 
 
 
 
 	Sinclairzhu	04:20:14 am	
 
yes
 
 
 
 
 	2999181A	04:20:42 am	
 
I think it is about UTC 9:20 now.

Have I missed the work shift for review for this day?
 
 
 
 
 	Sinclairzhu	04:21:36 am	
 
no sir
 
 
 
 
 	2999181A	04:22:51 am	
 
I will watch and make payment soon.

Thanks so much.

Let me wish you a all good day.
 
 
 
 
 
You rated our customer service as good
04:24:02 am	
 
 
 
 
 
You left the following comment: Thanks for the help.
04:24:12 am	
 
 
 
 
 
You archived the chat
04:24:58 am	
 
 
 
 
 
You left the following comment: Very productive.
04:25:20 am	
 
 
 
 	
Was the case resolved during the chat? Yes

Duration: 12m 12s
Chat started on: [https://jlcpcb.com/o...2997.1678299607-1420349117.1676504028](https://jlcpcb.com/order?_ga=2.20929080.1418382997.1678299607-1420349117.1676504028)
		
	
E-mail from [LiveChat](https://www.livechatinc.com/powered-by-livechat/?utm_source=chat_transcript&utm_medium=email&utm_campaign=jlcpcb.com)

LokiMetaSmith, 2023-03-16:
[image]
gap is still in the STL files.

ForrestErickson, 2023-03-17:
@LokiMetaSmith  see this LED stand off file: 
https://github.com/PubInv/general-alarm-device/blob/main/Hardware/Enclosure/U_Box_V105_GPAD_LED_Standoff_single.stl

ForrestErickson, 2023-03-17:
Regarding the  Hanging panel, Please see this file:
https://github.com/PubInv/general-alarm-device/blob/main/Hardware/Enclosure/BackPanel_Hanging-CompoundV2.stl

[image]

ForrestErickson, 2023-03-17:
@LokiMetaSmith 
We should review and remove old files. I did not feel comfortable removing files you created. I contemplated removing my files but wanted to wait till I knew I had success with the new files and then it slipped my mind.
I guess the arrival of parts will prove success and then removing old files can be the last act of this issue.

ForrestErickson, 2023-06-20:
Parts received from JLCPCB worked fine.

closed 2023-06-20 by ForrestErickson