PubInv/general-purpose-alarm-device #215: The GPAD_API should be updated to reflect the change of the buzzer in Rev 2.0 PCB

ForrestErickson, 2023-04-24:
Similar to Factory Test for Rev 2.0, Change the Buzzer Test from Tone to Simply High #214

The GPAD_API should be updated to reflect the change of the buzzer in Rev 2.0 PCB

ForrestErickson, 2023-07-08:
Adding @RobertLRead , @NULLbodyHERE  and @ForrestErickson  to this issue.

Capturing some recent email traffic here:

An email from Robert to NULLBodyHERE

 

> I actually thought of a programming task that you might really enjoy (as a DJ?!?), although I would prefer it is done before we publish this paper, which we are targeting for July 15th. 
>  
> I'm super busy as well, so I am trying to foist this off on you---which is a negative way of thinking about it---but still I think it will be fun.
>  
> Right now, the GPAD "shreiks". It is annoyingly loud. I don't actually remember if the volume changes with the alarm level.  The hardware you have may be our old-style hardware, it might be very quiet compared to what we have now.
>  
> I would like someone to design an "audio signature" that corresponds to each of the 5 alarm levels, so they can be distinguished by ear (even if you can't see the LEDs.) An example would be: 5 shreiks, then pause, for level 5: wee-wee-wee-wee-wee PAUSE wee-wee-wee-wee-wee-wee etc.
>  
> But, possibly there is a better way to do it-- like "wee-wee, wee-wee-wee" (long-long, short-short-short).
>  
> Of course, the sound of the alarm levels should get more annoying and attention-grabbing as the alarm level goes up. I don't think a steady tone is the most attention grabbing---if it is in a different room, a steady tone might be easier to ignore than a pattern.
>  
> What I'd like is for some to design these patterns---hopefully by testing with experience, and maybe doing some Google research on it. You could code it (it would require reading the clock and some logic), but I can code a pattern that you describe if you don't know how to program it.
>  
> Let me know what you think about this.
>

ForrestErickson, 2023-07-08:
@RobertLRead , @NULLbodyHERE 

Caution two GPAD hardware versions.

Please note there GPAD with PCB version 1, serial numbers 1-15 have a piezo transducer which can be driven by the tone() command at any frequency the tone() command supports but is not very loud.  If a moving coil speaker is attached to the connector (?J601 IIRC) then there is much move volume.

The GPAD Version 2 are serial number 16-30 and have a "buzzer" which makes a continuous tone which cannot be changed when the GPIO line to which the tone() command in our previous firmware (Version TBD???)  When you write at tone() to the buzzer the result is an intermodulation of the tone() frequency you set with the (again IIRC ) 2.4 KHz of the buzzer.  The result is not harmonically pleasant sound. The result is not as loud as simply setting the GPIO high and turning on the buzzer. 

So as you work with new firmware keep in  mind the target hardware for which you are developing and on which you are evaluating.

ForrestErickson, 2023-07-08:
An email from Robert Reed,

Objet : Significantly improved audio system....
 
Dear Team,
I just committed new GPAD main code which allows you to program "songs" corresponding to each of the 6 alarm levels. The number in the below is the "index" into the:

 const int BUZZER_LVL_FREQ_HZ[]= {0,128,256,512,1024,2048};

array to play at each 500-ms "tone" in the song. I don't use index 5 because at 2KHz it is (to my ear) to high pitched to be loud (though I have a high-frequency hearing loss.

You will notice I programmed some alternating "songs". I have attempted to do "urgency matching" --- that means as the alarm level increases, it should sound increasingly urgent, although that may be to some extent culturally dependent.

Seyi, I have sort of stepped on what I asked you to do, but there is still room for experimentation and improvement here (if you have a GPAD with our latest, louder buzzer.)  So hopefully I have not "jerked you around", but made it easier for you to experiment with, if you want to and have time.

I did this mostly so that in the paper I could say we have good audio cues. In fact, we had been simply using a steady tone of increasing frequency. That is arguably not very good at all.

const unsigned int NUM_NOTES = 20;
const int SONGS[][NUM_NOTES] = { { 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0},
 { 1,1,1,1,1,0,0,0,0,0,1,1,1,1,1,0,0,0,0,0},
 { 2,2,0,2,2,0,0,0,0,0,2,2,2,0,2,2,0,0,0,0},
 { 3,3,3,0,3,3,3,3,0,3,3,3,0,3,3,3,0,0,0,0},
 { 4,0,4,0,4,0,4,0,0,0,4,0,4,0,4,0,4,0,0,0},
 { 4,4,2,0,4,4,2,0,4,4,2,0,4,4,2,0,4,4,2,0}};

const unsigned LEN_OF_NOTE_MS = 500;

unsigned long start_of_song = 0;

ForrestErickson, 2023-07-08:
An email from NULLbodyHERE

Dear Robert,

Thank you for the additions you have made to this task. I should have communicated earlier that my plan was to finish it this weekend, after having recently retrieved my GPAD prototype (from my items currently sitting at my uncle’s place). I mistakenly forgot the power cable, so I resorted to the Wokwi simulated version. In any case, I’m grateful for your implementation!

I have made slight modification to the songs such that they are (in my opinion) more distinguishable from each other. Each level, except level 5, will include the level number in each phrase. For example, level 2 will issue a 2-note phrase, and level 4 will issue a 4-note phrase (nee-naw-nee-naw -break-). But also, the frequency of the phrase increases (to a certain extent) as the rest between each phrase decreases with rising alarm level. Level 5 is a constant two-tone alarm, giving it a distinct sound that is hard to ignore due to the lack of breaks. 

Hopefully my changes are appropriate. If not, please modify them to your liking. I’ll soon be more setup to work on this project a little bit more, as I’m moving some of my stuff in pretty soon. Thanks for your patience and all the best with the remainder of the paper/code changes.

ForrestErickson, 2023-07-08:
@RobertLRead ,
@NULLbodyHERE 

When you evaluate your work please make note on which hardware version you tested and which device serial number and any modifications.

ForrestErickson, 2023-07-08:
Regarding, " I resorted to the Wokwi simulated version. In any case, I’m grateful for your implementation!"

Can you share a link to the simulation here please?

NULLbodyHERE, 2023-07-10:
> Regarding, " I resorted to the Wokwi simulated version. In any case, I’m grateful for your implementation!"
> 
> Can you share a link to the simulation here please?

https://wokwi.com/projects/369869124017584129

RobertLRead, 2023-07-10:
Dear @NULLbodyHERE ,
I did a pull today and listened to all of Seyi's "songs". I like them better than the one's that I created. I therefore consider this mission to be accomplished! Good job. I'm closing this. We are now ready to finish the HardwareX paper.

closed 2023-07-10 by RobertLRead

ForrestErickson, 2023-07-11:
@RobertLRead 
What is the serial number of the GPAD on which you listened to the songs?

RobertLRead, 2023-07-11:
The case is black and has ears. Do you need me to open it up to find it on
the board?

ForrestErickson, 2023-07-11:
That GPAD is one of the five built into enclosures and therefor has the Buzzer and not the Piezo which is the important part for evaluation.  This reflects fully our Rev 2 design.  

Hurray!