PubInv/general-purpose-alarm-device #220: Gauging Enclosure Screw Holes, JLCPCB Nylon Enclosure of 202303

ForrestErickson, 2023-04-25:
## Notes on the JLCPCB Enclosure Screws.
### For the PCB mounting holes.
Using drill bits as gauges I find that a 7/64 " - 0.109 drill enters the holes to near but not quite to full depth.
A number 34 drill bit, 0.111" (2.82mm) enters the holes only about 1/8' to 3/16 with noticeable sliding resistance.

A #4 has no grip.
A #5 would have about 14/1000 interference.
A #6 would have about 27/1000 interference.

[image]

A metric M3 will have about 0.18mm or 7/1000" interference.   

[image]

ForrestErickson, 2023-04-25:
Before driving to purchase M3 screws, I tried some of my too long M3 screws with a nut on them to shorten them and found that they are too small.  Some of them slid down into the hole even under light pressure on the Philips head screw.

I finally used some 6-32 which I cut to length. They had good retention force and acceptable torque.

I ran out of the tiny screws for the side connection but used some screws I think are #2 machine screws on one unit.

LokiMetaSmith, 2023-04-25:
We will need a different BOM for hardware for different 3D printing
techniques.
Good to know!