PubInv/general-purpose-alarm-device #237: CH340 May Be Poor Choice for USB to UART Interface

ForrestErickson, 2023-06-21:
There is occasional traffic on Facebook groups regarding driver problems with the CH340 USB to UART interface chip and Windows.

Example: https://www.facebook.com/groups/1606743019578078/permalink/3377336785852017/?mibextid=Nif5oz

> I have a few ESP8266 CH340 boards that flash fine for a while with Windows 11, then after a few days i have to re-install the drivers because windows replaces the port driver with its own.
> I realised this after i threw a few boards away because of port errors.

Other users on that thread report problems too. 
Windows 11 is especially condemed.

ForrestErickson, 2023-06-21:
@RobertLRead 
@LokiMetaSmith 

Nothing is future proof but for any future design perhaps moving to the FTDI chip around which the UNO was developed is a good idea.

History, I chose the CH340 because they were used on some UNOs I have and they were available at JLCPCB at the time I checked and we were worried about availability.

LokiMetaSmith, 2023-07-05:
the fundamental problem is that USB to UART chips should have their internal device USB device string modified for the final device application to make it unique. 

Because diy has become production, we're seeing the logical conclusion of abusing the existing device designation.
The actual chip doesnt matter, but be it FTDI or CH340, if we were to put this into serious production, the USB to UART chip would need to be programmed with a unique ID and we would need to issue a signed firmware for integration into Microsoft's driver database.