PubInv/general-purpose-alarm-device #242: Changing baud connection rate resets the GPAD

RobertLRead, 2023-07-02:
While playing with the GPAD to try to make a nice photo figure for our paper, I discover that changing the baud rate connection (from the Arduino IDE 1.8.19) causes the GPAD to reset completely.

This is unexpected behavior. I think it should be reproduced on a PC, for example. It suspect it is an electrical problem; hopefully this is related to previous problems that Lee understands.

ForrestErickson, 2023-07-03:
@RobertLRead 

Disconnecting and reconnecting to the USB serial port causes a reset. This is characteristic of all UNOs by design. So I think this is expected behavior.  And by disconnecting and reconnecting I mean, for example,  closing and opening the Serial Terminal on the Arduino IDE. 

Please try a similar experiment on a standard UNO.

RobertLRead, 2023-07-03:
You are correct. I did not test, but found this discussion:

https://forum.arduino.cc/t/why-arduino-reset-on-baud-rate-change/383943

closed 2023-07-03 by RobertLRead