PubInv/general-purpose-alarm-device #245: HardwareX Paper, Descirption of Design based on the SCAD Reference Design

ForrestErickson, 2023-07-08:
From Lee's email of 
6/25/2023 11:14 AM

The design of the enclosure should give credit to the SCAD reference design  with which Lawrence started. Lawrence can you provide details?

ForrestErickson, 2023-07-08:
I have not seen traffic indicating that the enclosure design has been described and so I made this issue.

LokiMetaSmith, 2023-10-24:
from the source scad file

              -  FB Aka Heartman/Hearty 2016     -                   
              -   http://heartygfx.blogspot.com    -                  
              -       OpenScad Parametric Box      -                     
              -         CC BY-NC 3.0 License

ForrestErickson, 2023-10-24:
Lee added credits to https://github.com/PubInv/general-purpose-alarm-device/blob/main/Hardware/Enclosure/README.md

@RobertLRead IIRC you said you would add to the paper.

LokiMetaSmith, 2023-10-24:
I've pushed the changed to the paper, per commit 92c3afa, closing as complete

closed 2023-10-24 by LokiMetaSmith