PubInv/general-purpose-alarm-device #255: Fill in the expected values marked TBD in manuscript in the Electrical Tests section

RobertLRead, 2023-08-13:
Lee has documented things very thoroughly but there remands a 5 items marked in bold TBD s in the Electrical test area.

ForrestErickson, 2023-08-15:
> Assigned #255 https://github.com/PubInv/general-purpose-alarm-device/issues/255 to @ForrestErickson https://github.com/ForrestErickson.

Regarding, 
Fill in the expected values marked TBD in manuscript in the Electrical Tests section #255

Please provide a URL to the document in question.

RobertLRead, 2023-10-22:
@ForrestErickson 
https://github.com/PubInv/general-purpose-alarm-device/blob/main/doc/HardwareX%20GPAD%20Submission.odt

There is no only one TBD --- it is TBD ohms for on the line that begins "Power Jack". 

"Power Jack Measure resistance to ground at J101 center pin as greater than TBD ohms."

If you want to put it in this issue (the ohms), I will add to the paper

ForrestErickson, 2023-10-22:
I have updated and pushed the document. No more TBDs.

closed 2023-10-24 by RobertLRead