PubInv/general-purpose-alarm-device #263: Create proper Libraries that can be conveniently included...

RobertLRead, 2024-05-01:
I believe these files:

   GPAD_ALARM_API_SPI.cpp
    GPAD_ALARM_API_SPI.h
    alarm_api.h

should be moved into a separate library to make it as easy as possible to use a GPAD from a separate microcontroller.  This probably means creating a separate repo with a "library.json" file to fit into the Arduino example.

ForrestErickson, 2024-05-08:
I am not fully aware of what constitutes a "library" at the Github level, but would a separate "Controller" folder with in the GPAD project be sufficient?

RobertLRead, 2024-05-08:
No, there is a better way to do it, with a "library.json" file. We may
have to make a separate repo. I will find the documentation and do it fully.

I have done it before for the "PIRDS" system. If I include good
documentation hear, we can make it work.

ForrestErickson, 2024-05-08:
I have read this long ago but never tried to follow it enough to succeed:

https://docs.arduino.cc/learn/contributions/arduino-creating-library-guide/