PubInv/general-purpose-alarm-device #265: Change Enclosure for Moving Coil Speaker

ForrestErickson, 2024-08-11:
# Summary
Experiments with the Version 1 PCB with a push pull amplifier driving a moving coil speaker gave loud buzzer alarms at various frequencies.

A speaker from amazon was identified and a FreeCAD 3D model made by measurements taken from a real part. See file at: 
https://github.com/PubInv/general-purpose-alarm-device/tree/main/Hardware/Enclosure/Speaker2W
[image]

This may be the part Lee measured:
https://www.amazon.com/Speaker-Speakers-Compatible-Loudspeaker-Player/dp/B09YH5JZ1K/ref=sr_1_2?

cc: @nk25719 
@RobertLRead

nk25719, 2024-08-12:
@samoyin

samoyin, 2024-08-24:
## Enclosure for Moving Coil Speaker

```
translate([0,0,3.75]) cylinder (d1=13.80, d2=13.80, h=1.5);

color("LimeGreen") translate ([0,0,2.24]) cylinder (d1=26.070, d2=25.070, h=1.51);

color ("orangered")translate ([0,0,1.24]) cylinder (d1=26.630, d2=26, h=1.51);
difference(){
color("Steelblue") translate ([0,0,0.352]) cylinder (d1=28.980, d2=27.980, h=1.24);

translate([0,0,-1.5]) sphere(r=4); $fn=100;
}
difference(){
color ("plum") cylinder (d1=23.98, d2=23.98, h=3.352);

translate([0,0,-1.5]) sphere(r=4); $fn=100;
}
$fn=100;

translate ([0,0,0.5]) cylinder (r1=3, r2=4, h=0.5); $fn=100;

```

[image]
> Side View of the Speaker
[image]
> Bottom View of the Speaker

nk25719, 2024-08-24:
@ForrestErickson please review @samoyin 's work

[image]

> The measurements she followed are demonstrated here. Is this the correct representation of the measurements for the speaker we are using in the GPAD?

ForrestErickson, 2024-08-31:
Regarding, "@ForrestErickson please review @samoyin 's work"
Yes it looks correct. 
Everyone note that the dimensions were made by measuring a single part and are therefor with out a known tolerance.
The enclosure design must accommodate additional uncertainty. I would suggest something like 2 to 5 % might be a starting point.

CC: @nk25719

ForrestErickson, 2024-08-31:
@samoyin 
Please add the SCAD files for the speaker with a descriptive name into the repository.

The location: https://github.com/PubInv/general-purpose-alarm-device/tree/main/Hardware/Enclosure/Speaker2W might be good place.