PubInv/general-purpose-alarm-device #266: LED Spacer Cheight Undefined

nk25719, 2024-11-03:
What is the Cheight ? 
Ref file: U_Box_V105_General_Alarm_Device.scad
line: 266

LokiMetaSmith, 2024-11-03:
> @LokiMetaSmith <https://github.com/LokiMetaSmith>.
> <PubInv/general-purpose-alarm-device/issue/266/issue_event/15095380441@
> github.com>

Cheight is the height of the spacer.

for example in:

LedSpacer(1,PCBLength-27.94,15.24,5,2.54,Thick+FootHeight+PCBThick/2+.1-6,false);
//LED1

 Cheight is,   Thick+FootHeight+PCBThick/2+.1-6

which looks like it has a bunch of offsets and needs to be cleaned up.

ForrestErickson, 2024-11-04:
@LokiMetaSmith 
We asked a different question, Looking in the file we have not found where Cheight received an assignment or definition.

However,
I just got a bit smarter and copied the entire code into Notepad++ where I can do a find all and found this:

> [image]

So I see it is assigned from "Cyl height"

> [image]

However I can not find where "Cyl height" is assigned. 

## Bottom Line
I cannot answer the question, "What is the length of the LED Spacer as designed?"

LokiMetaSmith, 2024-11-04:
I'm a bit unsure of what problem you're ultimately trying to address. 

A brief overview,
The enclosure uses OpenScad, which is a procedural language reading functions(module in openscad lexicon) and variables.

I used on line 93
LED_Standoff    = 0;
LED_Standoff_Single    = 1; 

to control enabling and disabling the drawing of the LED_Standoff part.

If you search for LED_Standoff, you'll find a section where the drawing is built up, starting on lines 594 and 605

for example, on line 607 the module LedSpacer is called like thus:
  LedSpacer(1,PCBLength-27.94,15.24,5,2.54,Thick+FootHeight+PCBThick/2+.1-6,false);   

LedSpacer is defined on line 266
module LedSpacer(OnOff,Cx,Cy,Cdia,Cpitch,Cheight,Ccenter=false){ 

Cheight is a local variable only visible to the LedSpacer module and is unique to that module

Breaking things up a bit for readability
,Cx, , = PCBLength-27.94
Cy, = 15.24
Cdia, = 5
Cpitch, = 2.54
Cheight, = Thick+FootHeight+PCBThick/2+.1-6
Ccenter=false

Thick is the wall thickness of the case =  2
FootHeight is the height of the built in standoff feet = 15
PCBThick is the thickness of the pcb = 1.6

so Cheight = 2+15+1.6/2 + 0.1 - 6 
These variables were chosen because they adjusted the height so it didn't interfere with the LED's just barely poking out the top of the case. 

In best practice, the variable for LED_Standoff, should be defined in the global variable section. 
The LED Standoffs were a very late addition

ForrestErickson, 2024-11-04:
Regarding, "I'm a bit unsure of what problem you're ultimately trying to address."
We want the length of the LED_Standoff as used in the GPAD.

Looking again with new insight I see the FreeCAD file:  LED_Standoff_V104.FCStd
Which renders as:
[image]
Not what we want.

I see a file: U_Box_V105_GPAD_LED_Standoff_single.stl
[image]
for which I can not find a source code.