PubInv/general-purpose-alarm-device #267: Project Firmware\GPAD_API\GPAD_API.ino, Version of 20241104, Compile Error

ForrestErickson, 2024-11-04:
**Describe the bug**
_A clear and concise description of what the bug is._
Short Summary of the expected behavior:  Compile the GPAD firmware 
Short Summary of the buggy behavior:
Failed to compile.
[image]
Text: invalid conversion from 'uint8_t' {aka 'unsigned char'} to 'AlarmLevel' [-fpermissive]
 
 Details
Sketch: general-alarm-device\Firmware\GPAD_API\GPAD_API.ino

ForrestErickson, 2024-11-04:
The full verbose compiler message in this file:
[error.txt](https://github.com/user-attachments/files/17621455/error.txt)

RobertLRead, 2024-11-04:
> @RobertLRead <https://github.com/RobertLRead>.
> <PubInv/general-purpose-alarm-device/issue/267/issue_event/15107331247@
> github.com>

This is fixed in the latest code --- I suspect you are taking the GPAD code
from the GPAD repo, not the Krake repo.

ForrestErickson, 2024-11-04:
I confirm the Krake GPAD _API code compiles with out error.